Koenkk/zigbee2mqtt #24407: Sonoff ZBMINI R2 Maximum inching time

Kontendesigner, 2024-10-19:
### What happened?

I get an error, trying to put the inching tim on more than 127 seconds

### What did you expect to happen?

I expected to have a maximum of 3599,5 seconds (1 Hour)

### How to reproduce it (minimal and precise)

Every time I try tu set the inching time higher then 127 seconds

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240920

### Adapter

SONOFF ZBMINI R2

### Setup

plain on Proxmox 

### Debug log

2024-10-19 18:23:56z2m: Publish 'set' 'inching_control_set' to 'B Garage Steckdose außen' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0x7cc6b6fffef4e1ab/1 customClusterEwelink.protocolData({"data":[1,23,7,128,128,0,1200,4,0,0,1189]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"manufacturerCode":4742,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 1200)'

closed 2024-12-26 by github-actions[bot]

gacopl, 2024-12-31:
I also have this issue

FinnVonB, 2025-02-04:
same issue reported https://github.com/Koenkk/zigbee2mqtt/issues/25303