Koenkk/zigbee2mqtt #24409: [Feature request]: Aqara H1 Smart wall switch with property `click_mode` when `operation_mode: decoupled`

adam-kilis, 2024-10-19:
### Is your feature request related to a problem? Please describe

I've noticed that when I have Aqara H1 switch (WS-EUK02) in `operation_mode_left: decoupled` I observe delay trigger (light turn on). 

### Describe the solution you'd like

I want to have a control over `click_mode` property as remote device from family [WXKG15LM](https://www.zigbee2mqtt.io/devices/WXKG15LM.html#aqara-wxkg15lm), which allow to provide {"click_mode": "fast"} in order to not listen for multi click.[](url)

### Describe alternatives you've considered

I do not see alternative other than having `WXKG15LM` to have an control over "click_mode" property

### Additional context

https://www.zigbee2mqtt.io/devices/WS-EUK02.html#operation-mode-enum-left-endpoint
https://www.zigbee2mqtt.io/devices/WXKG15LM.html#click-mode-enum

andinger, 2024-10-21:
I‘m also interested in Controlling the Click Mode of my ws-euk04. If somebody could explain to me what changes are needed, i would provide zve necessary pull requests. Thx!

dziedzic99, 2024-11-25:
Unless you can recreate this behavior by setting something in the Aqara app using Aqara bridge, then this is likely due to internal software of the WS-EUK02 which doesn't support such feature.

closed 2025-02-01 by github-actions[bot]