Koenkk/zigbee2mqtt #24414: [New device support]: Tuya Smart ZigBee radiator valve

sebkr, 2024-10-20:
### Link

https://de.aliexpress.com/item/1005005978316733.html

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c138b01b585562","nwkAddr":4688,"manufId":4417,"manufName":"_TZE200_9xfjixap","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","modelId":"TS0601","manufacturerName":"_TZE200_9xfjixap","powerSource":3,"zclVersion":3,"appVersion":67,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1729411973340}

### Zigbee2MQTT version

1.40.2

### Comments

Manufacturer: _TZE200_9xfjixap is not recognised

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_9xfjixap',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing extended.
Please add the manufacturer _TZE200_9xfjixap to the main package like here: https://github.com/Koenkk/zigbee2mqtt/issues/16515

CraftingGab, 2024-10-23:
Same problem. The device is not recognized by the external converter. So i think something is wrong in the external conveter code.

mastersteve1, 2024-10-23:
no way to get it working. 

I also tried the external definitions but get still 'unsupported Device'

[image]

useahawk, 2024-10-24:
This works for me:
https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/main/avatto_me167.js

Change line 67 and 76 to your model, ie `_TZE200_9xfjixap`

CraftingGab, 2024-10-24:
@useahawk Thanks!  Works also for me!

Pichuhoehle, 2024-10-28:
@useahawk It worked for me too. Did you already put a pull request, so everyone can benefit from the support for the device?

Dome-Maurice, 2024-11-09:
@Sep21-2009 i set it up this up this way just now, and it works with better thermostat.

0verEngineer, 2024-11-14:
> This works for me: https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/main/avatto_me167.js
> 
> Change line 67 and 76 to your model, ie `_TZE200_9xfjixap`

This leads to the problem that i can set the Local temperature calibration only from -3 to +3, if i join it to a tuya gateway i can set from -10 to +10 in the Smart Life app.

Gnarfoz, 2024-11-21:
> This leads to the problem that i can set the Local temperature calibration only from -3 to +3, if i join it to a tuya gateway i can set from -10 to +10 in the Smart Life app.

You can adjust the limits on line 89:
https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/1ccbca29247db59d254d5b23f5c49c6f6e08b245/avatto_me167.js#L89

Dreaming-Codes, 2024-12-08:
This also works for _TZE204_ogx8u5z6: https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/main/avatto_me167.js

OneCreek, 2025-01-14:
for me the child_lock "Enables/disables physical input on the device" https://www.zigbee2mqtt.io/devices/ME167.html#child-lock-binary doesn't work properly... No difference if it is set to true or false. Anybody can confirm?

Gnarfoz, 2025-01-14:
Works just fine for me. 👍🏻

closed 2025-03-24 by github-actions[bot]