Koenkk/zigbee2mqtt #24417: Zigbee Dongle-E (Sonoff) does not start

clrmsc, 2024-10-20:
### What happened?

I can't connect to zigbee2mqtt.
config file:

configuration.yaml:
homeassistant: false
permit_join: true
mqtt:
base_topic: zigbee2mqtt
server: mqtt://localhost:1883
serial:
port: /dev/ttyACM0
baudrate: 115200
adapter: ember
rtscts: false
frontend:
port: 8081
advanced:
homeassistant_legacy_entity_attributes: false
legacy_api: false
legacy_availability_payload: false
device_options:
legacy: false

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl

### Adapter

SONOFF Zigbee Dongle-E

### Setup

Raspberry Pi 5 (64 bit OS)

### Debug log

> zigbee2mqtt@1.40.2 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2024-10-19 15:09:37] info:     z2m: Logging to console, file (filename: log.log)
[2024-10-19 15:09:37] info:     z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-19 15:09:37] info:     z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-19 15:09:38] info:     zh:ember: Using default stack config.
[2024-10-19 15:09:38] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-10-19 15:09:38] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-10-19 15:09:38] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 15:09:38] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-10-19 15:09:38] info:     zh:ember:uart:ash: Serial port opened
[2024-10-19 15:09:38] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 15:09:40] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 15:09:40] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 15:09:43] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 15:09:43] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 15:09:45] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 15:09:45] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 15:09:48] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 15:09:48] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 15:09:50] error:    z2m: Error while starting zigbee-herdsman
[2024-10-19 15:09:50] error:    z2m: Failed to start zigbee
[2024-10-19 15:09:50] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-19 15:09:50] error:    z2m: Exiting...
[2024-10-19 15:09:50] error:    z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1583:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:161:27)
    at start (/opt/zigbee2mqtt/index.js:154:5)

camlin, 2024-10-20:
For info, I am also using the Sonoff Dongle-E (the one marked Zigbee 3.0 USB Dongle Plus) and it is working fine. I am however on an x86 CPU (not ARM like the Raspberry Pi). But I would be surprised if that was the issue (many other people have used the Sonoff dongle on Raspberry Pi).

I am admittedly using a much simpler config file (and letting it decide what is best to use):
```
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.2.5
serial:
  port: /dev/ttyACM0
frontend:
  port: 8070
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
availability: true
```

And my start log is this:
```
info  2024-10-20 14:52:12: Logging to console and directory: '/app/data/log/2024-10-20.14-52-12' filename: log.txt
info  2024-10-20 14:52:13: Starting Zigbee2MQTT version 1.35.1 (commit #9eaaa0f)
info  2024-10-20 14:52:13: Starting zigbee-herdsman (0.30.0)
info  2024-10-20 14:52:16: zigbee-herdsman started (resumed)
info  2024-10-20 14:52:16: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
...
```

I have recently tried to use `1.40.2` but found it unstable MQTT wise (it kept reporting MQTT connections timeout), so I have admittedly reverted back to `1.35.1` but the ZigBee part/dongle had no issues with `1.40.2`

camlin, 2024-10-20:
Also, are you hinting at the fact it previously worked but suddenly stopped? If that's the case I would check if the user running the process is still the same (it will need to have read & write access on /dev/ttyACM0 and, if I remember, I did ran into some issues when I setup mine a while ago).

Nerivec, 2024-10-20:
Update your adapter to 7.4.x to use with `ember`.
https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl
You can flash it with:
- https://darkxst.github.io/silabs-firmware-builder/
- https://github.com/Nerivec/ember-zli

Then make sure your configuration (port) is correct. If possible, use the unique `/dev/serial/by-id/` path, to avoid issues with assignments for `tty`-type paths.

clrmsc, 2024-10-21:
> Update your adapter to 7.4.x to use with `ember`. https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl You can [...]

Thks! I flash it with SecureCRT 9.5 but now led no blinking.
and z2m started. can you say pls its work correct? 

Starting Zigbee2MQTT without watchdog.
[2024-10-21 05:50:20] info:     z2m: Logging to console, file (filename: log.log)
[2024-10-21 05:50:20] info:     z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-21 05:50:20] info:     z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-21 05:50:20] warning:  zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-10-21 05:50:23] info:     zh:ezsp:driv: Leaving current network and forming new network
[2024-10-21 05:50:24] info:     zh:ezsp:driv: Form network
[2024-10-21 05:50:26] info:     zh:controller: Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json'
[2024-10-21 05:50:26] info:     z2m: zigbee-herdsman started (reset)
[2024-10-21 05:50:26] info:     z2m: Coordinator firmware version: '{"meta":{"maintrel":"4 ","majorrel":"7","minorrel":"4","product":13,"revision":"7.4.4.0 build 0"},"type":"EZSP v13"}'
[2024-10-21 05:50:26] info:     z2m: Currently 0 devices are joined.
[2024-10-21 05:50:26] info:     z2m: Zigbee: disabling joining new devices.
[2024-10-21 05:50:26] info:     z2m: Connecting to MQTT server at mqtt://192.168.31.16
[2024-10-21 05:50:26] info:     z2m: Connected to MQTT server
[2024-10-21 05:50:26] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-10-21 05:50:26] info:     z2m: Started frontend on port 8070
[2024-10-21 05:50:26] info:     z2m: Zigbee2MQTT started!

camlin, 2024-10-21:
From the logs I would say it looks like it is now working correctly.

You just need to enable the join mode (temporarily, via the web-ui) so that you can add zigbee devices to be under its control (the exact procedure is device dependent but they usually have a button you need to press for 5 seconds or similar).

Nerivec, 2024-10-21:
Looks good indeed. 👍 

You should switch to `adapter: ember` before going any further, unless you have a good reason not to. `adapter: ezsp` is deprecated as indicated in your logs.
Then follow the docs to set things up properly:
https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config

clrmsc, 2024-10-23:
> Looks good indeed. 👍
>
> You should switch to `adapter: ember` before going any further, unless you have a good reason not to. `adapter: ezsp` is deprecated as indicated in your logs. Then follow the [...]

so strange, but in configuration.yaml:
  port: /dev/ttyACM0
  **adapter: ember**
  baudrate: 115200

Nerivec, 2024-10-23:
Make sure Z2M is indeed using the file you are editing. Based on your last logs, it should be `/opt/zigbee2mqtt/data/configuration.yaml` by default.

Here is an example "clean-slate network start" template (make sure to edit the settings as needed by your setup):
**This is a "clean-slate" template, to create a brand new network. It will force the adapter to form a new network, if any devices were previously joined they will need to be re-paired.**
```yaml
homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
serial:
  port: /dev/serial/by-id/my_usb_adapter_id
  adapter: ember
  baudrate: 115200
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
advanced:
  transmit_power: 20
  log_level: info
  log_output:
    - console
    - file
  timestamp_format: YYYY-MM-DD HH:mm:ss.SSS
  pan_id: GENERATE
  ext_pan_id: GENERATE
  network_key: GENERATE
  channel: 20
  last_seen: ISO_8601_local
```

camlin, 2024-10-26:
@clrmsc Is it now working for you? If it is please consider closing the ticket...

camlin, 2024-10-26:
BTW, I had a spare Sonoff Dongle-E (which I use for testing) and I have upgraded it to the 7.4.4 firmware and as soon as I did that I had to set the adapter in the config (i.e. `adapter: ember`) as otherwise Z2M would not start anymore (my previous config did not specify the adapter as that used to be the way).

And startup log looked fine:
```
[2024-10-26 14:16:51.886] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-26 14:16:52.028] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #unknown)
[2024-10-26 14:16:52.029] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-26 14:16:52.192] info: 	zh:ember: Using default stack config.
[2024-10-26 14:16:52.193] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-10-26 14:16:52.193] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-10-26 14:16:52.193] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-26 14:16:52.194] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-10-26 14:16:52.217] info: 	zh:ember:uart:ash: Serial port opened
[2024-10-26 14:16:52.217] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-26 14:16:53.316] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-10-26 14:16:53.345] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-10-26 14:16:53.345] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-10-26 14:16:53.361] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-10-26 14:16:53.368] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2024-10-26 14:16:53.535] info: 	zh:ember: [STACK STATUS] Network up.
[2024-10-26 14:16:53.570] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2024-10-26 14:16:53.584] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-10-26 14:16:53.599] info: 	z2m: zigbee-herdsman started (resumed)
```

clrmsc, 2024-12-28:
sorry guys, but it can't start anyway.

```
Starting Zigbee2MQTT without watchdog.
[2024-12-28 00:31:30] info:     z2m: Logging to console, file (filename: log.log)
[2024-12-28 00:31:30] info:     z2m: Starting Zigbee2MQTT version 1.42.0 (commit #bc6eccf)
[2024-12-28 00:31:30] info:     z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-28 00:31:30] info:     zh:ember: Using default stack config.
[2024-12-28 00:31:30] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-12-28 00:31:30] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-12-28 00:31:30] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-28 00:31:30] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-12-28 00:31:30] info:     zh:ember:uart:ash: Serial port opened
[2024-12-28 00:31:30] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-12-28 00:31:31] info:     zh:ember:uart:ash: ======== ASH connected ========
[2024-12-28 00:31:31] info:     zh:ember:uart:ash: ======== ASH started ========
[2024-12-28 00:31:31] info:     zh:ember:ezsp: ======== EZSP started ========
[2024-12-28 00:31:31] info:     zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-12-28 00:31:31] info:     zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2024-12-28 00:31:31] info:     zh:ember: [STACK STATUS] Network up.
[2024-12-28 00:31:31] info:     zh:ember: [INIT TC] Adapter network does not match config. Leaving network...
[2024-12-28 00:31:32] info:     zh:ember: [STACK STATUS] Network down.
[2024-12-28 00:31:32] info:     zh:ember: [INIT TC] No valid backup found.
[2024-12-28 00:31:32] info:     zh:ember: [INIT TC] Forming from config.
[2024-12-28 00:31:32] info:     zh:ember: [INIT FORM] Forming new network with: {"panId":6754,"extendedPanId":[221,221,221,221,221,221,221,221],"radioTxPower":5,"radioChannel":11,"joinMethod":0,"nwkManagerId":0,"nwkUpdateId":0,"channels":134215680}
[2024-12-28 00:31:32] info:     zh:ember: [STACK STATUS] Network up.
[2024-12-28 00:31:32] info:     zh:ember: [INIT FORM] New network formed!
[2024-12-28 00:31:32] info:     zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-12-28 00:31:32] error:    z2m: Error while starting zigbee-herdsman
[2024-12-28 00:31:32] error:    z2m: Failed to start zigbee
[2024-12-28 00:31:32] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-28 00:31:32] error:    z2m: Exiting...
[2024-12-28 00:31:32] error:    z2m: Error: EACCES: permission denied, rename '/opt/zigbee2mqtt/data/database.db.tmp' -> '/opt/zigbee2mqtt/data/database.db'
    at Object.renameSync (node:fs:1032:11)
    at Database.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/database.ts:120:12)
    at Controller.databaseSave (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:381:23)
    at Controller.backup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:385:14)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:187:24)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:161:27)
    at start (/opt/zigbee2mqtt/index.js:154:5)
```

toine512, 2025-01-01:
Your permissions are borked. Most likely you've edited `database.db` as root and now Z2M can't r/w it.
This is out of scope for an Issue, go to the forum Discussions.

closed 2025-03-11 by github-actions[bot]