Koenkk/zigbee2mqtt #24420: zigbee2mqtt/bridge/devices does not seem to be retained

camlin, 2024-10-20:
### What happened?

I am trying to get (via MQTT) the list of devices currently connected to z2m. As I understand it, this is done by subscribing to the `zigbee2mqtt/bridge/devices` topic, which should be retained (See https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-devices and https://github.com/Koenkk/zigbee2mqtt/discussions/11294).

But, when I subscribe to that topic (or `zigbee2mqtt/#` to be sure), I currently get nothing on that topic (or I get the other retains):
```
MQTT: Subscription to zigbee2mqtt/bridge/devices successful
MQTT: Subscription to zigbee2mqtt/# successful
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/Main-Door/availability, payload=18byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/Chambre/availability, payload=18byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/Bureau/availability, payload=18byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/Cuisine/availability, payload=18byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/Bibliotheque/availability, payload=18byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/Garage/availability, payload=18byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/bridge/extensions, payload=2byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/bridge/groups, payload=2byte, qos=AT_MOST_ONCE, retain=true}
MQTT: Received message: MqttPublish{topic=zigbee2mqtt/bridge/state, payload=18byte, qos=AT_MOST_ONCE, retain=true}
```

### What did you expect to happen?

When connecting to the MQTT server, I was expecting a publish message to occur on the `zigbee2mqtt/bridge/devices` topic that would have included the device list as described in https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-devices

For information, my MQTT server also indicates that `zigbee2mqtt/bridge/devices` is not in the retained list:
```
MQTT Peer Status

Client name	mqttjs_da826cab	
Time	1:40:13	
Subs	1	
Retains	9	
Reconns	0	
Messages allocated	10	
Remote IP	192.168.2.11	
Tx messages	234	
Rx messages	215	
Will set	true
Clean	true
Subscription: zigbee2mqtt/#

Retained: zigbee2mqtt/bridge/state {"state":"online"}
Retained: zigbee2mqtt/bridge/groups []
Retained: zigbee2mqtt/bridge/extensions []
Retained: zigbee2mqtt/Garage/availability {"state":"online"}
Retained: zigbee2mqtt/Bibliotheque/availability {"state":"online"}
Retained: zigbee2mqtt/Cuisine/availability {"state":"online"}
Retained: zigbee2mqtt/Bureau/availability {"state":"online"}
Retained: zigbee2mqtt/Chambre/availability {"state":"online"}
Retained: zigbee2mqtt/Main-Door/availability {"state":"online"}

Will: zigbee2mqtt/bridge/state {"state":"offline"}
```

So I wonder if I need to set a special configuration setting to enable this...

### How to reproduce it (minimal and precise)

Use a MQTT client to connect to `zigbee2mqtt/bridge/devices` and check if you get the last retained state.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle-E

### Setup

Plain with default Zigbee2MQTT docker

camlin, 2024-10-26:
I suspect I will end up just listening to `zigbee2mqtt/#` but that's not always ideal (causing traffic for no reason, for example I don't need to listen to Z2M's log channel).

I am only trying to get the list of devices and monitor it so that I know what I need to subscribe to.

I can't be the only one trying to do that? And the docs point to the fact it should be possible no? (i.e. https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-devices ).

I will perhaps look at the source code at some point to try and find out the correct answer but that feels a bit strange...

Koenkk, 2024-10-26:
bridge/devices is retained, maybe something wrong with your broker config?

camlin, 2024-10-26:
Ok, thanks for confirming it should be retained. What is strange for me is that other things are being retained, but not that one 🤔

The MQTT broker is a pretty simple one that is embedded in a FireBrick FB2900 router. There is nothing in its config at the moment that would affect this (basically just setting the basics of port and creds). 

But I think I just found out what could be happening: I have many devices and I think the broker is discarding the publish to `zigbee2mqtt/bridge/devices` because the message is too big...

I am in the process of validating my theory but I am expecting to close the ticket shortly. 

But thanks again for confirming I was at least reading the docs correctly and that I was trying to check the correct topic.

camlin, 2024-10-26:
Ok, confirmed. The issue was caused by my MQTT broker limiting the size of payloads to 3,000 bytes (which makes sense tbh as a lot of IoT devices can't cope with big payloads).

On my main system I have 6 devices (which is not a lot) and that was causing a `zigbee2mqtt/bridge/devices` publish in excess of 25k (so about 10 times the limit set by the broker).

Using a test system with just one device (and the same MQTT broker) worked fine (i.e. I was able to subscribe get notified of device changes).

Also worked fine if I use Mosquito as the broker...

I learned something today :-)

Thanks for your help @Koenkk

closed 2024-10-26 by camlin

camlin, 2024-10-26:
But it does make me wonder if, long term, the payload should not be optimised. Because it is true that someone having many devices (in the several hundreds is not unheard of) could end up with very big payload on the `zigbee2mqtt/bridge/devices` topic.

Perhaps it should should only list the devices IDs and then let the client subscribe to a dedicated topic (for the device?) if they want more information? 🤔

Koenkk, 2024-10-27:
Until there are some actual problems with this, I don't see a reason to optimise (as it will only complicate things)