Koenkk/zigbee2mqtt #24422: device: Namron 4512750/4512751 issues on firmware 3.10

SanderBlom, 2024-10-20:
### What happened?

This summer, I ordered 10-15 dimmers from Namron. Initially, I received only 4 devices, with the remaining ones arriving in October. The first batch, which came with firmware version 3.09, performed flawlessly—allowing me to turn the lights on and off and adjust the brightness with immediate response. However, the final batch arrived with firmware version 3.10, which caused issues. The lights would take a few seconds to turn on, and after about 10-15 seconds, I encountered an error in Z2M.

**Turning the light ON**
``` 
Publish 'set' 'brightness' to 'lys-soverom-2' failed: 'Error: ZCL command 0x84b4dbfffec4d232/1 genLevelCtrl.moveToLevelWithOnOff({"level":120,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 21842 - 1 - 100 - 8 - 11 after 10000ms)'
``` 

**Dimming the light**

```
Publish 'set' 'brightness' to 'lys-soverom-2' failed: 'Error: ZCL command 0x84b4dbfffec4d232/1 genLevelCtrl.moveToLevelWithOnOff({"level":82,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 21842 - 1 - 104 - 8 - 11 after 10000ms)'
``` 

**Turning the light OFF**
```
Publish 'set' 'state' to 'lys-soverom-2' failed: 'Error: ZCL command 0x84b4dbfffec4d232/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 21842 - 1 - 105 - 6 - 11 after 10000ms)'
```

So all the new dimmers with the `3.10` firmware have this issue that turning them on or off takes a long time (2-10 sec). 

After contacting the distrubutor in Norway about the difference in the firmware,  I got this resposne back (from device manufcation in china): 

  > Changes v3.09 > v3.10
  Implementation of Min, Max and start-up level of lightness.
   Clusters for dimmer settings Min,Max, Start-Up light levels
  Implement in the dimmer fw in the dimmer MCU the chosen ZB Clusters respectively:
  (The chosen ZB Cluster is from the zigbee standard for Ballast, not Light control. Since there is no defined ZB Cluster for these settings in the ZB standard library)
   Cluster 0x301 (Ballast Configuration)
   Attribute 0x001 (Ballast settings)
  Attributes set 0x0010 (Min level setting) (6 repeated short push on the self reset switch)
  0x0011 (Max level setting) (8 repeated short push on the self reset switch)
  0x0012 (Power On level) (5 repeated short push on the self reset switch)
  These chosen attributes may be implemented in the ZB instructions in the MCU.
  We think it can be tested by sending commands to the UART Interface on the CPU ( RX/TX) directly via cable.
  Because we still don’t have any App to test on. It will be available in September.

### What did you expect to happen?

I expected that they would respond the same as the other devices with the `3.09` firmware 

### How to reproduce it (minimal and precise)

Toggle the light on/off and check the error logs about 10-20 seconds after changing the dimmer's state.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle

### Setup

add-on on Home Assistant

### Debug log

 2024-10-20 20:43:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lys-fairy', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":30,"energy":0.09,"led_disabled_night":null,"linkquality":60,"overload_protection":null,"power":9.2,"power_outage_memory":null,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:43:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":4.7,"led_disabled_night":null,"linkquality":132,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:43:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lys-fairy', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":30,"energy":0.09,"led_disabled_night":null,"linkquality":57,"overload_protection":null,"power":9.2,"power_outage_memory":null,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:43:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":4.7,"led_disabled_night":null,"linkquality":132,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:44:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lys-fairy', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":30,"energy":0.09,"led_disabled_night":null,"linkquality":60,"overload_protection":null,"power":9.2,"power_outage_memory":null,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:44:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":4.7,"led_disabled_night":null,"linkquality":135,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
[... 48 lines cut ...]
error 2024-10-20 20:45:32z2m: Publish 'set' 'state' to 'lys-soverom-2' failed: 'Error: ZCL command 0x84b4dbfffec4d232/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 21842 - 1 - 117 - 6 - 11 after 10000ms)'
info 2024-10-20 20:45:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lys-fairy', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":30,"energy":0.09,"led_disabled_night":null,"linkquality":126,"overload_protection":null,"power":9.2,"power_outage_memory":null,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:45:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":4.7,"led_disabled_night":null,"linkquality":132,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:45:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lys-fairy', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":30,"energy":0.09,"led_disabled_night":null,"linkquality":126,"overload_protection":null,"power":9.2,"power_outage_memory":null,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:45:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":4.7,"led_disabled_night":null,"linkquality":135,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:45:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lys-fairy', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":30,"energy":0.09,"led_disabled_night":null,"linkquality":126,"overload_protection":null,"power":9.2,"power_outage_memory":null,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"id [... cut ...] on":45,"state":"idle"},"update_available":null,"voltage":null}'
info 2024-10-20 20:46:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":4.7,"led_disabled_night":null,"linkquality":132,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'

SanderBlom, 2024-10-29:
> I have the same issue with my Namron 5412751. I just got it two days ago. First command is working, and then I get 7-15sec delay on the next command. In the debug log I can see that it reports Installed_version20 / latest_version10. 
> 
> info 2024-10-24 20:40:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/NamronDimmer2', payload '{"brightness":30,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"installed_version":20,"latest_version":10,"state":"idle"},"update_available":null}'

I would advise you to try to contact namron and hopefully they can come with a fix. They told me that if multiple people experienced the same issue they could potentially look into it.

fyksen, 2024-10-30:
I have two batches of the `4512751`.
The ones I ordered first came with the firmware version:  ` 0x03096530 ` , the other `0x00000014`

I'm using ZHA, not zigbee2mqtt. But seems like the problems are the same.
With the first one I have this issue (sent mail to Namron today): ` 0x03096530 `

---

When the light is on and an automation turns it off, everything works as expected. However, if the light is already off and I run an automation to turn it off again, then try to turn the light back on using the dimmer button, it doesn't turn on to the minimum set value. Instead, it turns on to the absolute lowest setting possible, resulting in that the light doesn't come on at all. I've set the minimum value by pressing the dimmer six times, but this setting isn't respected when Home Assistant turns off the light after it's already off.

---

The second seems to have the same problems as you. generally being unresponsive, when send multiple commands, and not send status back to ZHA.

Sorry for piling on in this issue tracker, just wanted to add my findings.

SanderBlom, 2024-11-24:
@fyksen Did you get any response back from Namron?

SanderBlom, 2024-11-25:
> > @fyksen Did you get any response back from Namron?
> 
> Just that they did not support z2m and ZHA. But that they might support it in the future, but right now it was too much work. I asked for firmwares, so I could downgrade them to the same 3.09, but haven't heard back.
> 
> I really hope that others also contact Namron about this. It is really easy, just send them a mail on [post@namron.com](mailto:post@namron.com)

I received a similar response, but they mentioned they'd consider it if multiple people reported the same issue. While they don’t need to officially support it, they should at least publish the firmware and Zigbee cluster specifications for each device. This would make it much easier for us to add support on our own.

Kilill, 2024-11-29:
Same here Namron 4512750 with FW 3.10, have mailed Namron, currently no response though...

SanderBlom, 2024-12-12:
I now got some information from Namron saying that they are working with the hardware manufacturer to solve this issue. 

 https://www.hjemmeautomasjon.no/forums/topic/12466-namron-dimmer-45127514512750-problemer/#comment-116857

Kilill, 2024-12-13:
Just a side note, Namron are "Elektroimportøren"'s (https://www.elektroimportoren.no) own product name, contacting/sending mail to "Namrons" advertised mail address (post@namron.com) is an exercise in futility as it seems that mail address is a black hole. 
It is better to contact Elektroimportørn directly through their contact form.

Also, Elektroimportøren seems to publish most of their firmwares on the product page (under the Documentation section), but for 4512750 / 4512751 there are ofcourse none.

But they do have for the alternate 1402769. I havent tried that one, since i also have a Z-Wave net i ended up exchanging all these zigbee dimmers for the same zwave version , and they work just fine....

fyksen, 2024-12-13:
I am aware that Namron is Elektroimportørens brand, but I think it still is good practice to contact the right company email.
I got answer very quickly from post@namron.com. If you tried contacting that email in the last weeks, I suspect that they have gotten a lot of email (from people that look up this github issue, and also I posted on the Home Assistant Norge Facebook group), and maybe they have gotten tired of it, and stopped answering. Atleast it proved the point that there are alot of people looking for solutions. IDK, just my toughts.

Kilill, 2024-12-14:
In principle i agree with you but there are several reasons why i decided to contact Elektorimportøren

1. Absolutely no response from the Namron email address
2.  I actually bought the dimmers from Elektroimportøren.
3. Namron is a 100% owned brand name of Elektroimportøren.
4. When calling Namron's service phone, they answer with "Elektorimportøren"
5. The recommendation  from that phone call was that mailing the Namron email might, eventually, reach Elektroimportøren complaint department anyway and it was better to send complaints directly to the Elektroimportøren

SanderBlom, 2025-01-24:
For information Namron has now just released a new firmware that fixes the current issue. I did a manual upgrade yesterday to test the new firmware and it currently looks like it work as expected. 

I have created a new [PR](https://github.com/Koenkk/zigbee-OTA/pull/659) to add the `.ota` file to the repository that should resolve this issue.

closed 2025-01-24 by SanderBlom

IDmedia, 2025-06-10:
I also emailed them since I have a lot of issues with my Namron 1402769 constantly throwing errors.
It's a bit early to know if this fixes the issue for me, but I thought it would be nice to post it here for completeness. 

> Hei,
> 
> "Hi Pål,
> 
> **Can you change the max report interval to 420**?" 
> 
> Det er feedbacken vi har fått, kan du prøve å endre til det?
> 
> Ha en fin dag videre! 

[image]

Alpacay, 2025-06-16:
> I also emailed them since I have a lot of issues with my Namron 1402769 constantly throwing errors. It seems to help, but it's too early to know if this actually fixes the issue or if this is just a coincidence, but I thought it would be nice to post it here for completeness....
> 
> **Edit:** The setting doesn't seem to work as the errors are back after 24 hours.
> 
> > Hei,
> > "Hi Pål,
> > **Can you change the max report interval to 420**?"
> > Det er feedbacken vi har fått, kan du prøve å endre til det?
> > Ha en fin dag videre!
> 
> [image]

@IDmedia Hello，I have the  Namron 1402769 device, but i can't see these errors.  Can you tell me what's your Zigbee2MQTT version, Adapter firmware version,  Adapter.

SanderBlom, 2025-12-17:
@IDmedia I have deployed just the V2 version of the dimmer you described (4512750/4512751) and they have all  been working great! Maybe one dimmer disconnects from the zigbee network after a few months but that is about the only problem I have experienced with them.

IDmedia, 2025-12-31:
For completeness, here's a video of the issue I have with Namron: https://youtu.be/mcw_-vqUpus

I'm going to request an update from Namron in the end of January and if a new OTA isn't available by the end of February then I'm going to follow my wife's demand to throw out these garbage dimmers. It's enormously frustrating when the lights don't work properly during the holidays and half the devices go offline from time to time.

IDmedia, 2026-01-12:
Super excited news!

```
Hello again!
We currently have an FW version (1.18) in testing with us that fixes all these issues. It must go through a testing process with the supplier to ensure that all functions work as intended and that it does not disrupt any existing functionality.
If all tests are successful, we will release an update in the next few days.
```

But versioning seems hard... website says v35, Z2M says 3.09 and they mention 1.18....

Yesterday I got pissed enough that I brought another SLZB dongle and moved all of the Namron dimmers to their own network so they can spam each other instead of my other devices. That solved all my stability issues, but obviously I'm not a big fan of having to run multiple Zigbee-networks.