Koenkk/zigbee2mqtt #24441: HA on Proxmox - 1.40.2 and/or Ember - Z2M restart and rolling shutter go-stop-go

nerthazrim, 2024-10-22:
### What happened?

When I restart the Z2M addon in HA, my shutters all do a quick move-stop-move, as if the Zigbee network has been opened for pairing (they do the same if I hit "Permit join (all)" in Z2M webUI.
I noticed the issue because sometimes, my Z2M addon stops because of a timeout with the controller during a Proxmox VM backup. See here:
```
[2024-10-22 02:08:09] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-10-22 02:08:09] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-10-22 02:08:09] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-10-22 02:08:09] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-10-22 02:08:09] error: 	z2m: Adapter disconnected, stopping
```

The crash during PVE backup is a minor issue. What's not minor for me is that when Z2M restarts, the shutters all move... If you check the time of the crash in the logs above, I'll let you imagine the jumpscare it provoked during the night.
Now, the funny thing is that I thought it was because of Ember (I moved from EZSP recently) but it's not, at least not only:
```
"OK" setup: Z2M 1.40.1 + ezsp driver
"Not OK" setup: Z2M 1.40.2 + ezsp driver
"Not OK" setup: Z2M 1.40.1 + ember driver
"Not OK" setup: Z2M 1.40.2 + ember driver
```

"OK" means the shutters are not moving when I restart. "Not OK" they move.

Does anyone has any idea of what is provoking this? Is the Zigbee network really opening at Z2M restart on latest version and/or with Ember? Any way to prevent this?

Additional question (but I can create a separate issue if need be): is there a way to increase the timeout to avoid Z2M restarting during my PVE backups?

### What did you expect to happen?

Shutters not moving (= Zigbee network not opening?) when Z2M restarts

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

 7.4.4.0 build 0

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

HA VM in Proxmox Virtual Environment

jan0o0sch, 2024-10-24:
I am having the same problem ... also under PROXMOX.
Is it possible, that PROXMOX has anything to do with it?

nerthazrim, 2024-10-25:
Hi,

I've seen your issue yes but it did not seem to be the same issue.

First, I would like to understand why my shutters are moving only with a specific version of Z2M and/or with ember but not with ezsp.
I would like to know if it's intended or a bug somehow.

Second, my timeout only happens during VM backups, never otherwise so it does not seem to be a stress issue or anything similar to your problem (except for the timeout itself) but maybe I'm missing the link.
For this point, I would like to know if the timeout is configurable, as I could not find the information in the extensive doc :(

nerthazrim, 2024-11-04:
Hi,

I'm hoping not to be too much of a pain in the ass @Koenkk, just to you let you know the behavior is the same with 1.41.0 :(
If I upgrade to 1.41.0 or/and if I move from ezsp to ember, my roller shutters behave as if the Zigbee network was being opened on "Permit all" when Z2M starts and stops.
Any idea how I could change this behavior on Z2M, if that's actually the case? Or give you more information to understand why my shutters are suddenly moving?

Additionally, any idea how I could increase the ember timeout with the USB adapter?

Thanks in advance

nerthazrim, 2024-12-02:
Closing this one and I'll open 2 separate issues if the problems remain with v2.0.0+

closed 2024-12-02 by nerthazrim