Koenkk/zigbee2mqtt #24443: Ikea LED1624G9 does not pair

herbod-nl, 2024-10-22:
### What happened?

Ikea LED1624G9 does not pair

### What did you expect to happen?

Ikea LED1624G9 to pair

### How to reproduce it (minimal and precise)

connected the lamp, zigbee2mqtt found it immediately but gave an error message. Lamp reset, off and on 6 times. same result.

error 2024-10-22 20:51:06zh:controller: Interview failed for '0x90fd9ffffef3612a with error 'Error: Interview failed because can not get node descriptor ('0x90fd9ffffef3612a')'
error 2024-10-22 20:51:06z2m: Failed to interview '0x90fd9ffffef3612a', device has not successfully been paired

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

Firmware: 0.6.2

### Adapter

 Zig Star WT32-ETH01

### Setup

add on home assistant , in proxmox

herbod-nl, 2024-10-22:
With pairing the bulb was 15 cm next to the coordinator

tjakobi, 2024-10-25:
I placed the light into another socket next to the original socket - pairing immediately worked again. This issues seems to have been be triggered by issues with the socket/power delivery to the bulb?

closed 2025-01-01 by github-actions[bot]