Koenkk/zigbee2mqtt #24445: Exception configuring Sonoff TRVZB thermostatic valves: value is not a number (using Phoscon ConBee II)

porphyr, 2024-10-22:
### What happened?

I bought a bundle of 4 Sonoff TRVZB valves. Of these, I could only pair 1 successfully with Zigbee2mqtt.
For the others, I receive the error message 
"z2m: Exception while calling fromZigbee converter: Value is not a number, got undefined(undefined)}" twice.
It says (also twice) "z2m: Successfully configured '0x38398ffffe067f9a'", and the device is listed as paired, however the "network icon" disappears on the valve shortly after, and there is no communication between valve and controller anymore.
This sounds somewhat similar to #22448, but unfortunately, enabling debug logging does not fix the issue in my case, it is persistent and repeatable with the 3 affected valves

### What did you expect to happen?

Valve is successfully paired

### How to reproduce it (minimal and precise)

Join a Sonoff TRVZB Thermostatic valve and the issue appears: confirmed on three of four valves.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Add-on in Home Assistant on Lenovo ThinkCentre M910q (x86-64))

porphyr, 2024-10-22:
[log.log](https://github.com/user-attachments/files/17481915/log.log)
adding logfile

Ricc68, 2024-11-06:
I confirm that I don't see the issue anymore.
I think one of the most effective actions I took was to buy an **USB2** extension cable and bring the dongle 1.5m far from WiFi router and from there on my ZigBee network communications became rock solid.
In addition I suggest to update to the latest version of z2m.

closed 2025-01-14 by github-actions[bot]