Koenkk/zigbee2mqtt #24447: [Feature request]: Improvements to PJ-1203A - Reset Energy Sum and Channel Multiplier

tdashmike, 2024-10-23:
### Is your feature request related to a problem? Please describe

There is currently no way to reset the total energy consumption and use a CT clamp with a higher current rating.

### Describe the solution you'd like

An option to reset the total energy consumption and a multiplier for each channel for different CT clamps.

### Describe alternatives you've considered

You can probably reset the energy sum in the Tuya app but I did not install it yet.

### Additional context

@schauveau Is this something you can help with? Thanks!

schauveau, 2024-10-23:
You can reset the Energy (and all datapoints) by resetting the device (a very long press?) and pair it again.   

At one point, I had the calibrations datapoints working but I did not find them very useful because the multipliers are applied independently of each others. For example, the Current calibration was not affecting power (i.e. calibration of I and U is ignored when computing P=UI) and the Power calibration was not affecting Energy. 
Even worse, Energy calibration is applied to the final result and not during the "accumulation" meaning that changing the calibration introduces a huge jumps in the reported value. I still have a few huge nonsensical energy values in my HomeAssistant database because of that. 

Anyways, if you want to try it yourself then look at the commented code in my old repo
https://github.com/schauveau/z2m_PJ-1203A/blob/main/PJ-1203A-v3.js

You will notice in my code that the calibration ranges are limited to [ 0.5 , 2 ]. 
```.withValueMin(0.5).withValueMax(2.0)```

I do not remember if this is an actual limitation of the device or something I introduced myself.

If the calibration range is really limited to `[0.5,2]` then another approach could be to replace 
```  [106, 'energy_a', tuya.valueConverter.divideBy100 ] ```
by something like 
```  [106, 'energy_a', tuya.valueConverterBasic.divideBy(N) ] ```
where N is either 300 or 33.3 

The easiest way to experiment is probably to copy and modify my latest external converter [PJ-1203A-v4.js](https://github.com/schauveau/z2m_PJ-1203A) . I did not try it since the code was accepted in Z2M 6 months ago but it should hopefully still work.

schauveau, 2024-10-23:
Also, it may be easier to scale the values in the final application. If you are using HomeAssistant, create a copy of each entity scaled by the proper factor.

digitlength, 2024-10-26:
> (a very long press?)

For those trying this, for me it wouldn't reset after holding for 1 minute, but would after holding for 2 minutes.

tdashmike, 2024-11-18:
Wanted to post the configuration I'm using in Home Assistant to get an accurate measurement. Values are a tad low compared to what the power company is reporting but for my purposes it's good enough.

```
template:
  - sensor:
      - name: "Instantaneous Power Usage"
        unique_id: "bbbbbbbbbbbaaaaaaaaa"
        device_class: power
        state_class: measurement
        unit_of_measurement: "W"
        state: "{{ (states('sensor.energy_monitor_1_current_a')|float * states('sensor.energy_monitor_1_power_factor_a')|float + states('sensor.energy_monitor_1_current_b')|float * states('sensor.energy_monitor_1_power_factor_b')|float) * 3 / 100 * states('sensor.energy_monitor_1_voltage')|float }}"
```

```
sensor:
  - platform: integration
    name: "Energy Consumption"
    unique_id: "aaaaaaaaaaaaaaaaaabbbbbbbbbbbbbb"
    source: sensor.instantaneous_power_usage
    method: left
    #method: trapezoidal
    unit_time: h
    unit_prefix: k
    max_sub_interval:
      minutes: 5
```

closed 2025-01-25 by github-actions[bot]