Koenkk/zigbee2mqtt #24452: Smart Socket Elivco LSPA9 switches of randomly

bachmarc, 2024-10-23:
### What happened?

HI, 
I bought 10 Elivco LSPA9 sockets described [here](https://www.zigbee2mqtt.io/devices/LSPA9.html#elivco-lspa9) in docu.
I run no automations or scripts and just used the sockets for power monitoring.

I observed that the socket of our fridge turned off at night and stayed off for 16 hours until I got aware.

HA Logbook shows it becomes unavailable from time to time but keeps  last status after reconnect.
At  00:51:55 it switched off randomly. Nobody was in the room, no physical or virtual switch pressed.

```
17:21:42 S013 Tiefkühltruhe Oma became unknown
17:22:05 S013 Tiefkühltruhe Oma turned on
17:23:16 S013 Tiefkühltruhe Oma became unknown
17:23:45 S013 Tiefkühltruhe Oma turned on
=> 00:51:55 S013 Tiefkühltruhe Oma turned off
```
after that the "restored status" was defined as off and the following 16 hours it stayed off even if connection was lost. A kind of works as designed but based on a random "switch off"

```
Zigbee2MQTT version
    [1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    EmberZNet

Coordinator revision
    7.4.1 [GA]

Coordinator IEEE Address
    0x8c65a3fffe5485e4

Frontend version
    0.7.4

zigbee-herdsman-converters version
    20.21.0

zigbee-herdsman version
    2.1.3

Stats

        Total 21
        By device type
        Router: 19
        End devices: 2
        By power source
        Mains (single phase): 19
        Battery: 2
        By vendor
        _TZ3000_gvn91tmx: 10
        _TZ3000_okaz9tjs: 9
        _TZ3000_lf56vpxj: 1
        _TZE204_upagmta9: 1
        By model
        TS011F: 19
        TS0202: 1
        TS0601: 1
```

In the "About" of Z2M I see "TS011F: 19."
I bought Elivco LSPA9 on purpose because I saw that the older Tuya TS011F_plug_1, I use,  were marked as unreliable in docu... 
Are the all the same? 

I tried OTA ... 
for the Elivco LSPA9 it fails in two ways:

`2024-10-23 11:52:37z2m: Failed to check if update available for 'S010 Audio' (Device didn't respond to OTA request)`

or 

`2024-10-23 11:53:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/S005 Trockner Oma', payload '{"child_lock":"UNLOCK","countdown":0,"current":1.66,"energy":3.27,"indicator_mode":"off/on","last_seen":"2024-10-23T11:53:39+02:00","linkquality":88,"power":331,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228}'`

=> a bit strange because OTA UI shows Firmware 0122052017

by the way:
for TS011F_plug_1 it crashes the adapter...

```
2024-10-23 11:45:34zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
2024-10-23 11:45:34zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
2024-10-23 11:45:34zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
2024-10-23 11:45:34zh:ember: Adapter fatal error: HOST_FATAL_ERROR
2024-10-23 11:45:42zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=235 Len=7].
2024-10-23 11:46:04z2m: Adapter disconnected, stopping
```

### What did you expect to happen?

I would expect that the sockets do not switch off randomly
Elivco LSPA9 seems to be TS011F and therefore it should be marked as "randomly switching off" like TS011F_plug_1
I woul expect that OTA gives a clear feedback and is able to update (while I am unsure if that fixes the random issue)

### How to reproduce it (minimal and precise)

That is difficult as it happens from time to time randomly.
Loglevel is currently info, don't know if "debug" will cause a data drama as it has to run for days to reproduce

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.1 [GA]

### Adapter

ZBDongle-E 3.0 USB Dongle Plus,EFR32MG21 + CH9102F/CP2102N Zigbee USB-Stick EFR32MG21 Coordinator

### Setup

Z2M as addon in Home assistant

yannpub, 2024-11-14:
Same here.
I haven't been able to find any pattern or root cause, so I built automations that monitor these plugs.
In case the device shows off more than a couple of minutes, the automation switch them back on automatically.

joel, 2024-11-14:
Same issue with ZHA

joel, 2024-11-27:
Is there a way to update the firmware on those devices? They do not seem to offer OTA. On the other hand, is there a way to deactivate the ZigBee router ability? I believe the router table becomes too large, which might be the issue.

yannpub, 2024-11-27:
@davidclaverie I do not think tension peaks are the reason. If it was from the grid, then they would switch off at the same time, and I haven't been able to identify peaks. And it does not seem related to the appliance on it (which I thought at the beginning), as I have some plugs switching off with only a router or a water valve plugged in, thus consumption <10W without peak.

davidclaverie, 2024-11-27:
What is the solution. i have about 10 plugs.....

joel, 2024-11-27:
@davidclaverie I have about 15, and I started to replace them with the [THIRDREALITY Zigbee Smart Plug](https://3reality.com/product/smart-plug-gen2-with-energy-monitoring/), which offer OTA:

[image]

[[image]](https://www.amazon.es/dp/B0CRKC6GTS)

yannpub, 2024-11-27:
@joel I have seen on elektrokoda that apparently they are based on a ZT2S module. So maybe it is possible to get or generate a new firmware via the Tuya developper account.

Deadmor48, 2024-12-02:
Same issue. Can fix only by reboot Zigbee2MQTT addon. It works about 1 hour after reboot. 
Who can advice me how to add watchdog to monitor status of that plug? -)

yannpub, 2024-12-03:
> Same issue. Can fix only by reboot Zigbee2MQTT addon. It works about 1 hour after reboot. Who can advice me how to add watchdog to monitor status of that plug? -)

I have automation rules in HA that switch them on if they appear OFF for a couple of minutes. But obviously that works only in you use them for energy monitoring, but not as switches.

davidclaverie, 2024-12-03:
Oui je vois mais cela veut dire aussi que si par exemple c'est sur une
machine a laver, sécher .. le programme de la machine peut se remettre à
Zéro. Si c'est pour un computer il peut s'éteindre sans avoir sauvegarder
son programme ou couper une visioconférence avec un client  par exemple

-- 
__________________________
David CLAVERIE​
Mobile / WhatsApp: +241 (0)66 26 25 00 <+241%20066%2026%2025%2000>
P Pensez a l'environnement avant d imprimer cet email . Please consider the
environment before printing this email
Les informations contenues dans ce message sont confidentielles et à
l'attention exclusive de son (ou ses) destinataire(s). Si vous n'êtes pas
destinataire de ce message, sachez que toute utilisation édition,
modification ou diffusion de ce message est strictement interdite. Si vous
avez reçu ce message par erreur, merci de nous en informer immédiatement
par téléphone ou email et de le supprimer de votre système.

Deadmor48, 2024-12-03:
> I have automation rules in HA that switch them on if they appear OFF for a couple of minutes. But obviously that works only in you use them for energy monitoring, but not as switches.

I'm using that switches now for turn ON/OFF outdoor new year's lights... waiting for the solution and looking for new switches)

davidclaverie, 2024-12-03:
OK i see.
do you test/see IKEA switch INSPELNING?

-- 
__________________________
David CLAVERIE​
Mobile / WhatsApp: +241 (0)66 26 25 00 <+241%20066%2026%2025%2000>
P Pensez a l'environnement avant d imprimer cet email . Please consider the
environment before printing this email
Les informations contenues dans ce message sont confidentielles et à
l'attention exclusive de son (ou ses) destinataire(s). Si vous n'êtes pas
destinataire de ce message, sachez que toute utilisation édition,
modification ou diffusion de ce message est strictement interdite. Si vous
avez reçu ce message par erreur, merci de nous en informer immédiatement
par téléphone ou email et de le supprimer de votre système.

Deadmor48, 2024-12-07:
> […](#)
> __________________________ David CLAVERIE​Mobile / WhatsApp: +241 (0)66 26 25 00 <+241%20066%2026%2025%2000> P Pensez a l'environnement avant d imprimer cet email . Please consider the [...]

No, i didn't test..

przemoc86, 2025-01-11:
Hello, any progress guys - yesterday one of the plugs turned off 3 times, second one 4 or 5 times. I'm just adding automations to turn on automatically plugs that should work all the time...

Deadmor48, 2025-01-11:
> Hello, any progress guys - yesterday one of the plugs turned off 3 times, second one 4 or 5 times. I'm just adding automations to turn on automatically plugs that should work all the time...

Unfortunately no any progress, I've just added node (in Node-Red)to check state of the plug after switch on/off and restart Z2M if state is wrong...

przemoc86, 2025-01-12:
> Unfortunately no any progress, I've just added node (in Node-Red)to check state of the plug after switch on/off and restart Z2M if state is wrong...

Don't tell me more :) Thankfully I'm able to switch them on without Z2M restart
Last 6 hours, 6 times switched off :/

[image]

yannpub, 2025-01-12:
> OK i see. do you test/see IKEA switch INSPELNING? 

Yes. Looks stable, didn't have any issue at all so far, and the power monitoring looks more accurate than these LSPA9 plugs

yannpub, 2025-01-15:
I just created a PR to update the Notes section of the device in the documentation with the following:

/|\ Important Information /|\
Some of these devices (if not all) randomly switch off.
Recommendation is not to use these for critical appliances as freezers or fridges, nor appliances which would be stopped during a programs like washing machines.
Only solution found so far is to monitor them and setup some automations to restart these when their status is off for a certain duration.
As of Jan 2025, no OTA update is available.

closed 2025-03-25 by github-actions[bot]