Koenkk/zigbee2mqtt #24453: Sonoff SNZB-02D

mrochette01, 2024-10-23:
### What happened?

I installed a first Sonoff "temperature and humidity sensor" (SNZB-02D). In general, inclusion went well. However, I got 2 errors, one in the log and the other in the frontend. These errors are still present despite several attempts at installation.
[image]

   

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

SMLIGHT SLZB-06

### Setup

HA with Raspberry Pi 4  rpi4-64

### Debug log

[2024-10-23 06:05:33] error: 	z2m: Failed to configure 'Temp: salon_2', attempt 1 (Error: Bind 0x44e2f8fffe17b916/1 genPowerCfg from '0x00124b0024c16ebc/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:403:47
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:311:16)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:30)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:111:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:157:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13))

maximvd, 2024-10-23:
This might be related to https://github.com/Koenkk/zigbee2mqtt/issues/24025

apydo, 2024-11-06:
It may be related to #24025. And several people added comments about SNZB-02D not displaying power source even on 1.41.0. However, #24025 is closed now. I'm not sure @Koenkk saw these comments.

In my case, there is no power source displayed for all the SNZB-02D thermometers with IEEE adresses starting with 0x44e2 (a batch of devices purchased recently).
I, of course, tried the solution suggested by @Koenkk and updated device config and restarted Z2M with no change.

BTW, another batch of SNZB-02D with IEEE adresses starting with 0x881a always worked correctly and shows power source with no problem for several months.

[image]

acommuni, 2024-11-24:
I have the same issue. I you need some help to tshoot, let me know.

Yasounet, 2024-12-01:
It appears to be fixed in 1.42.0. You have to make sure that device is awake (I've just pressed physical button on the back of the unit) and then reconfigure it using yellow button in Z2M gui. 
[image]

The same procedure did not work with previous 1.41.0 version of Z2M.

acommuni, 2024-12-01:
I do the same as @Yasounet and it works with last Z2M version