Koenkk/zigbee2mqtt #24456: Profalux Remote Repeatedly Disconnects from Zigbee Network After Successful Pairing

cjournois, 2024-10-23:
### What happened?

I'm attempting to connect a Profalux remote control (it resembles the NB102, though I'm unsure of the exact model). The remote joins the network initially, but then disconnects shortly after (as indicated by the logs).

The Profalux remote (NB102) successfully pairs with the Zigbee network but repeatedly disconnects shortly after. Logs indicate a possible database issue with the error message: DatabaseEntry with ID '38' does not exist.

### What did you expect to happen?

To establish connection with the remote.

### How to reproduce it (minimal and precise)

1. Permit to join (all)
2. Start the paring mode on the remote, after reset (5 press of R button on the back, wait a few seconds, then press the stop button)

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.4

### Adapter

EmberZNet

### Setup

RPI 4

### Debug log

 2024-10-23 21:31:04z2m: Successfully changed options
info 2024-10-23 21:31:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"0f7iz-2"}'
info 2024-10-23 21:31:07zh:controller: Succesfully interviewed '0x20918a00000ed8b7'
info 2024-10-23 21:31:07z2m: Successfully interviewed '0x20918a00000ed8b7', device has successfully been paired
info 2024-10-23 21:31:07z2m: Device '0x20918a00000ed8b7' is supported, identified as: Profalux Cover remote (NB102)
info 2024-10-23 21:31:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Cover remote","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NB102","options":[],"supports_ota":false,"vendor":"Profalux"},"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"successful","supported":true},"type":"device_interview"}'
warning 2024-10-23 21:31:07z2m: Device '0x20918a00000ed8b7' left the network
info 2024-10-23 21:31:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_leave"}'
info 2024-10-23 21:31:10zh:controller: Interview for '0x20918a00000ed8b7' started
info 2024-10-23 21:31:10z2m: Device '0x20918a00000ed8b7' joined
info 2024-10-23 21:31:10z2m: Starting interview of '0x20918a00000ed8b7'
info 2024-10-23 21:31:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_joined"}'
info 2024-10-23 21:31:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"started"},"type":"device_interview"}'
info 2024-10-23 21:31:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_announce"}'
info 2024-10-23 21:31:18zh:controller: Succesfully interviewed '0x20918a00000ed8b7'
info 2024-10-23 21:31:18z2m: Successfully interviewed '0x20918a00000ed8b7', device has successfully been paired
info 2024-10-23 21:31:18z2m: Device '0x20918a00000ed8b7' is supported, identified as: Profalux Cover remote (NB102)
info 2024-10-23 21:31:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Cover remote","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NB102","options":[],"supports_ota":false,"vendor":"Profalux"},"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"successful","supported":true},"type":"device_interview"}'
warning 2024-10-23 21:31:23z2m: Device '0x20918a00000ed8b7' left the network
info 2024-10-23 21:31:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_leave"}'
info 2024-10-23 21:31:31zh:controller: Interview for '0x20918a00000ed8b7' started
info 2024-10-23 21:31:31z2m: Device '0x20918a00000ed8b7' joined
info 2024-10-23 21:31:31z2m: Starting interview of '0x20918a00000ed8b7'
info 2024-10-23 21:31:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_joined"}'
info 2024-10-23 21:31:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"started"},"type":"device_interview"}'
info 2024-10-23 21:31:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_announce"}'
info 2024-10-23 21:31:33zh:controller: Succesfully interviewed '0x20918a00000ed8b7'
info 2024-10-23 21:31:33z2m: Successfully interviewed '0x20918a00000ed8b7', device has successfully been paired
info 2024-10-23 21:31:33z2m: Device '0x20918a00000ed8b7' is supported, identified as: Profalux Cover remote (NB102)
info 2024-10-23 21:31:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Cover remote","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NB102","options":[],"supports_ota":false,"vendor":"Profalux"},"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"successful","supported":true},"type":"device_interview"}'
warning 2024-10-23 21:31:44z2m: Device '0x20918a00000ed8b7' left the network
info 2024-10-23 21:31:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_leave"}'
info 2024-10-23 21:31:46zh:controller: Interview for '0x20918a00000ed8b7' started
info 2024-10-23 21:31:46z2m: Device '0x20918a00000ed8b7' joined
info 2024-10-23 21:31:47z2m: Starting interview of '0x20918a00000ed8b7'
info 2024-10-23 21:31:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_joined"}'
info 2024-10-23 21:31:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"started"},"type":"device_interview"}'
[... 2 lines cut ...]
warning 2024-10-23 21:31:59z2m: Device '0x20918a00000ed8b7' left the network
info 2024-10-23 21:32:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_leave"}'
info 2024-10-23 21:32:02zh:controller: Interview for '0x20918a00000ed8b7' started
info 2024-10-23 21:32:02z2m: Device '0x20918a00000ed8b7' joined
info 2024-10-23 21:32:02z2m: Starting interview of '0x20918a00000ed8b7'
info 2024-10-23 21:32:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_joined"}'
info 2024-10-23 21:32:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7","status":"started"},"type":"device_interview"}'
info 2024-10-23 21:32:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_announce"}'
info 2024-10-23 21:32:05zh:controller: Succesfully interviewed '0x20918a00000ed8b7'
info 2024-10-23 21:32:05z2m: Successfully interviewed '0x20918a00000ed8b7', device has successfully been paired
info 2024-10-23 21:32:05z2m: Device '0x20918a00000ed8b7' is supported, identified as: Profalux Cover remote (NB102)
[... cut ...]
info 2024-10-23 21:32:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_announce"}'
warning 2024-10-23 21:32:33z2m: Device '0x20918a00000ed8b7' left the network
info 2024-10-23 21:32:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918a00000ed8b7","ieee_address":"0x20918a00000ed8b7"},"type":"device_leave"}'
error 2024-10-23 21:32:35zh:controller: Interview failed for '0x20918a00000ed8b7 with error 'Error: DatabaseEntry with ID '38' does not exist'

Iroamstaff, 2024-10-29:
A mon avis tu a un moteur en zigbee 3.0 et que la télécommande est en zigbee 1.3.
Ca pose problème, beaucoup personnes sont dans la même situation.

cjournois, 2024-10-31:
> A mon avis tu a un moteur en zigbee 3.0 et que la télécommande est en zigbee 1.3. Ca pose problème, beaucoup personnes sont dans la même situation.

Bonjour,

Cela est frustrant, car j’ai réussi à appairer 3 de mes 5 volets (qui, je suppose, ont été installés en même temps, car commandés ensemble) simplement avec un appui sur F + HAUT. Pour ceux-là, ma télécommande fonctionne directement avec le volet (bindée avant l’appairage), et je peux les contrôler via z2m. Par contre, impossible d’appairer les deux derniers volets, même après un reset moteur (changement de fil, puis attente de la troisième vague de mouvements avant de débrancher).

J’ai vu dans un autre post que tu avais réussi à appairer tes volets. As-tu rencontré des difficultés particulières, et arrives-tu aujourd’hui à contrôler tous tes volets avec z2m ?

cjournois, 2024-10-31:
I'm closing the issue because I believe it's related to the device itself, not to Z2M.

Iroamstaff, 2024-11-08:
Je sais pas si tu y ai arriver mais en gros si tes volets sont de 2021, profalux a utiliser des télécommande en z1.2 et volets en z.3 et sur un même achat certains sont ok d'autre non. Si c'est le cas impossible de faire entrer la télécommande avec le volet elle finit toujours par quitter le réseau. Mais normalement les volets entre dans le réseau et ils fonctionnent. La solution c'est d'utiliser une télécommande et de faire une automatisation. 
Si c'est le volet que tu n'arrives pas a faire entrer dans le réseau c'est que la manipulation n'est pas la bonne