Koenkk/zigbee2mqtt #24460: [Feature request]: Allow for a rolling OTA update

dajomas, 2024-10-24:
### Is your feature request related to a problem? Please describe

No, there is no problem related to this request

### Describe the solution you'd like

when many devices have an update it would be useful to be able to do a rolling update to avoid overloading the Zigbee network and having to click update many times.

### Describe alternatives you've considered

I have not considered any alternatives

### Additional context

for instance, I have 10 lightbulbs at my dining table of the same type and they always have updates together, to avoid having to start 10 updates after eachother manually, requiring 2 clicks each, it would be better to have a button that will starts a process to update all the updatable devices one after another.

Koenkk, 2024-10-24:
I already started working this some time ago in https://github.com/Koenkk/zigbee2mqtt/tree/feat/ota-update-check-all, some other things came in between but might continue on it in the future.

WoodenDuke, 2025-03-02:
I'm a little sad that this was closed as "not planned" - I would be very, very happy about such a feature...

It would be a great option if you could configure the maximum number of simultaneous updates in the Z2M settings or have an option to "process updates one after the other". If Z2M would process the updates in sequence or with the maximum number, it would certainly save many users a lot of waiting.

For example, I have 13 TRVZB thermostats in use. Since each update takes about 50 minutes and more than 1 update at the same time often causes problems for me, I have to start an update every hour for a whole day. But if I go this route, the updates always work flawlessly for me.

Or does anyone have an idea for an alternative solution?

lawrencedudley, 2026-09-26:
Very much a stop-gap until something gets merged natively to handle queued OTAs but I've quickly popped together an add-on for Home Assistant that does what you're looking for: https://github.com/parallax/z2m-auto-update

It's running fine on the two HA instances I use, any feedback/bugs would be appreciated!