Koenkk/zigbee2mqtt #24465: Namron Zigbee LED Dimmer 4512751

Konstabel, 2024-10-24:
### What happened?

The device is listed as supported, and it is no problem to pair and use the device. The problem is that after the fist command is sent to the device, it gets very unresponsive. It takes between 7-15 sec to turn the dimmer on/off or adjust brightness. If I wait 1-2 minutes after last command is sent, the dimmer is snappy on first command, then on second/third command it gets very unresponsive again. It also reports that current firmware on dimmer is 20, but last firmware online is 10. 

### What did you expect to happen?

That the dimmer is fast to respond to commands and that no error is reported when sending commands. Matching firmware on dimmer and "online". 

### How to reproduce it (minimal and precise)

Happens every time I use  the dimmer (Except for the first command). 

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

EmberZNet 7.4.2 [GA]

### Adapter

Sonoff dongle -E

### Setup

Home assistant OS 13.2 on LenovoPC

### Debug log

error 2024-10-24 20:40:10z2m: Publish 'set' 'brightness' to 'NamronDimmer2' failed: 'Error: ZCL command 0x0ceff6fffe8fc5fb/1 genLevelCtrl.moveToLevelWithOnOff({"level":31,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":36725,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":115},"zclSequence":20,"commandIdentifier":11} timed out after 10000ms)'
info 2024-10-24 20:40:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/NamronDimmer2', payload '{"brightness":30,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"installed_version":20,"latest_version":10,"state":"idle"},"update_available":null}'
[log.log](https://github.com/user-attachments/files/17511462/log.log)

SanderBlom, 2024-11-25:
This is a duplicate of same issue listed here: https://github.com/Koenkk/zigbee2mqtt/issues/24422

SanderBlom, 2024-11-25:
@Konstabel As there is a bunch of people having issues with latest batch that shipped with the FW version 3.10 i would advice you to reach out to Namron and ask for a fix as this  may put pressure on them to come out with a solution as many people are affected by this.

closed 2025-02-01 by github-actions[bot]