Koenkk/zigbee2mqtt #24477: [New device support]: Support for ESP32-H2 and C6 devices with Arduino Code

MeisterQ, 2024-10-25:
### Link

https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples

### Database entry

--

### Zigbee2MQTT version

--

### Comments

Since arduino-esp32 has releaseed the new 3.0.7.R, new custom zigbee devices are available.

https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples

You can now flash any ESP32-H2 or C6 with Codes like dimmable light, switch, temperature Sensor etc.

Please add support for this devices.

They can also be a coordinator, router or end device.

### External definition

```shell
--
```

### What does/doesn't work with the external definition?

--

petrkr, 2024-11-13:
Those actually work out-of-box. Or Color light. That I tested...

I need somehow now to send/read raw messages from custom attributes

MeisterQ, 2024-11-13:
> Those actually work out-of-box. Or Color light. That I tested...
> 
> I need somehow now to send/read raw messages from custom attributes

Ive tested it on IObroker Zigbee Adapter which is using z2m. Couldnt get it work

closed 2025-01-20 by github-actions[bot]