Koenkk/zigbee2mqtt #24482: bht_002  MQTT zhc:legacy:tz:moes_bht_002: Existing program state not set.

dudeden, 2024-10-25:
### What happened?

When setting the schedule in Z2M on my thermostat I get the error zhc:legacy:tz:moes_bht_002: Existing program state not set. Specifying the schedule parameters either through manual input or through the selection buttons does not give a result.

### What did you expect to happen?

Saving entered values

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

CC2652P

### Setup

HomeAssistant

achim-70, 2024-11-01:
I have the same problem. Also currently on version 1.40.2 but I can't confirm that the problem only exists since this version as I didn't change these settings since more than a year.

dudeden, 2024-11-01:
Solution. Add a thermostat via TUYA zigbee gateway. Set a schedule via the TUYA application. After that, return the thermostat back to Z2M. Now changing the schedule via Z2M works. This seems to be a solution, but it would be great if it could be done without this "initialization via gateway"

istroganov, 2024-11-07:
Have some problem and havn't TUYA zigbee gateway :(

masminl, 2024-11-24:
After a power outage I had to re-pair some Zigbee devices including my BHT-002.

Same issue,  I can't program the thermostat via Z2M anymore. Purchasing a TUYA gateway just for this doesn't seem to be the solution. For now since it's winter I'm stuck with having to program the thermostat on the device itself argg.

achim-70, 2024-11-27:
My current problem was a wrong "Local temperature calibration". I found a way to set it via the display. It was not necessary to hold for 8 seconds, the setting appears already after 2 seconds. The temperature calibration is "Temperature compensation" in the table below.

[image]

closed 2025-02-03 by github-actions[bot]