Koenkk/zigbee2mqtt #24485: [New device support]: Phillips Hue Centris (4 spots white)

frest-de, 2024-10-25:
### Link

https://www.amazon.com/-/de/dp/B0CYZDM8GM/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1AAFDPPA4TETU&dib=eyJ2IjoiMSJ9.fbTtnLZ-y1zGQheei88bhDiex6xVjrrhVJyMMWmaMqfAqtiD_iPaezjoyIN1986Bk72GScB1Lo4L5sDpX6Lwzyci0cmFnjY4KW4x6jGinYeH-6es4jNXl1X89YgXXaGJV-JmlvTx05CK7kp-kdaXo9XyMXdizb750Vs8FwBAux7uXemFMsntG5SEkr97PsTY29p7m9imjrsnWvzkmrK9a9WyI9XxqSZoYhGgeacoH6Wlxc6up-1u2DRhQ649mqwjUIBTpk6RRUwdpqN7d-BT4YcAZo7tGa8YdvzzbuB8iM4.1wmZOB-lQIaykPGyfWxFNDl1pplAsrUH0mFW_68olqQ&dib_tag=se&keywords=hue%2Bcentris%2B4&qid=1729885006&sprefix=hue%2Bcentris%2B4%2Caps%2C185&sr=8-1&th=1

### Database entry

{"id":103,"type":"Router","ieeeAddr":"0x001788010e7fab62","nwkAddr":65273,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003808901_01","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"hwVersion":0,"zclVersion":2}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":26599,"currentY":25697,"colorTemperature":286}},"genLevelCtrl":{"attributes":{"currentLevel":157}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1729883751097}

### Zigbee2MQTT version

1.40.2

### Comments

Similar to the issue https://github.com/Koenkk/zigbee2mqtt/discussions/12173 and https://github.com/Koenkk/zigbee2mqtt/issues/20215, the model number seems to have changed.

Its:
929003808901_01
929003808901_02
929003808901_03
929003808901_04
929003808901_05

929003808901_02 - 929003808901_05 are GU10 White and color ambiance bulb, just with individual model numbers.

### External definition

```shell
unclear
```

### What does/doesn't work with the external definition?

unclear

Koenkk, 2024-10-26:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `929003808901.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

frest-de, 2024-10-26:
Thanks for your work!
I added the image.

closed 2024-10-27 by Koenkk