Koenkk/zigbee2mqtt #24486: Light entities seems to turn off, but in reality they stay on....

pauloon, 2024-10-25:
### What happened?

It's happening very frequently with my home system that you turn off some lights, either by doing it manually on a dashboard, or via a Node-Red automation, and everything shows as the command was accepted, the entity indicates it's off, the device on Z2M shows as off, but the light itself stays on.

This is happening randomly to many of my lights. I repair then and it seems to fix the issue, but after a while it starts again.

In the print below you can se the presence sensor detected the room was empty, and this sends a command to turn off the group with all the lights in this room:

[image]

The light shows up as off:

[image]

The Z2M shows the device as off:

[image]

And when you check the light, it is still on:

[image]

Then, if I resend the command to turn off the group lights, or only this light specifically, and it works (it turns off).

Can you, please, check?

### What did you expect to happen?

The light should turn off correctly as per command.

### How to reproduce it (minimal and precise)

I'm not able to reproduce it exactly. It looks random and it has happened to most of my lights, either the simple "switch" normal ones (that I convert to light using a helper), or the dimmable ones (that already show up as a light entity).

### Zigbee2MQTT version

 1.40.2-dev commit: c09aeec

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

I use HAOS installed in a Intel NUC, and the Z2M standard add-on. Recently I was testing the EDGE to see if there was any solution.

### Debug log

[log.log](https://github.com/user-attachments/files/17527975/log.log)
[log1.log](https://github.com/user-attachments/files/17527977/log1.log)
[log2.log](https://github.com/user-attachments/files/17527978/log2.log)

pauloon, 2024-10-29:
All of these devices have optimistic mode active. I'm not sure if there is a bug or a problem with that...

Duncank, 2024-11-02:
I'm seeing something similar, where after no motion is detected the lights do turn of, but home assistant and z2m keep showing that they are on. After motion is detected again, the light wont turn on since HA already thinks it is on.

Started happening after the latest z2m update, mostly with TS110E dimmers (_TZ3210_3mpwqzuu)

pauloon, 2024-11-03:
> I'm seeing something similar, where after no motion is detected the lights do turn off, but home assistant and z2m keep showing that they are on. After motion is detected again, the light wont turn on since HA already thinks it is on.
> 
> Started happening after the latest z2m update, mostly with TS110E dimmers (_TZ3210_3mpwqzuu)

Interesting. That happened to me too, but I didn't give much importance since 90% of times is the situation I described. Lights are commanded to be turned off, the entity shows it as off, but the light is actually still on. In my case here is happening to all kinds of light switches, normal and dimmable ones.

Duncank, 2024-11-05:
I've reverted to 1.40.2-1 last night and so far this problem hasn't reappeared.

aaronbeekay, 2024-11-20:
This is happening for me as well, with both ZHA and z2m, on two different TS110E (`_TZ3210_3mpwqzuu`) devices.

I haven't been able to do much fault isolation yet, besides switching them from ZHA to z2m. I'm on z2m 1.41.0.

pauloon, 2024-11-20:
> This is happening for me as well, with both ZHA and z2m, on two different TS110E (`_TZ3210_3mpwqzuu`) devices.
> 
> I haven't been able to do much fault isolation yet, besides switching them from ZHA to z2m. I'm on z2m 1.41.0.

For me here it did not happen anymore. I'm not sure which updated seems to have "fixed" that.

pauloon, 2025-01-22:
Now with the 2.0 update, this same problem started happening again.
Anyone else here?

Duncank, 2025-01-22:
I tried the upgrade again from 1.40.2-1 to the latest version 2.0, and this indeed is still not working correctly.

Using a simple light.turn_on action with a `brightness` or `brightness_pct`, the light does appear to turn on and go to the correct brightness in both home assistant and zigbee2mqtt, but the physical light itselfs just stays off.

Adapter: Sonoff 
Model: TS110E_2gang_2
Zigbee Model: TS110E
Zigbee Manufacturer: _TZ3210_3mpwqzuu

State before:
```
{
    "brightness": 25,
    "brightness_l1": 152,
    "brightness_l2": 255,
    "linkquality": 21,
    "max_brightness": 255,
    "max_brightness_l1": 61,
    "max_brightness_l2": 254,
    "min_brightness": 4,
    "min_brightness_l1": 1,
    "power_on_behavior": "previous",
    "power_on_behavior_l1": "previous",
    "power_on_behavior_l2": "previous",
    "state_l1": "OFF",
    "state_l2": "OFF",
    "switch_type_l1": "momentary",
    "switch_type_l2": "toggle"
}
```

State after turn_on command from home assistant:
```diff
{
    "brightness": 25,
    "brightness_l1": 152,
    "brightness_l2": 255,
    "linkquality": 21,
    "max_brightness": 255,
    "max_brightness_l1": 61,
    "max_brightness_l2": 254,
    "min_brightness": 4,
    "min_brightness_l1": 1,
    "power_on_behavior": "previous",
    "power_on_behavior_l1": "previous",
    "power_on_behavior_l2": "previous",
+    "state_l1": "ON",
    "state_l2": "OFF",
    "switch_type_l1": "momentary",
    "switch_type_l2": "toggle"
}
```

light remains off. I found [this commit](https://github.com/Koenkk/zigbee-herdsman-converters/commit/7e504f2e80beaf25b19372c3ea41585db5b9c1af#diff-7ae23d3c13c2d7f7e37106590684c791aba8093f8be813b7c74a4f1484017c3b) which seems to fix a similar problem for the 1 channel version of this dimmer. Maybe a similar thing is happening here?

I'm happy to provide more debug info.

Duncank, 2025-03-24:
This is still happening. (bump to remove the stale label)

bernardesarthur, 2025-05-05:
All of you using Sonoff Dongle P (CC2652P) in this case?

aaronbeekay, 2025-05-05:
@bernardesarthur I’m using SLZB-06M (Silicon Labs EFR32MG21 radio).

pauloon, 2025-05-05:
> All of you using Sonoff Dongle P (CC2652P) in this case?

I'm using Sonoff Dongle P.

Issue seemed to be fixed in past, but this week it started happening again after last update: 2.3.0
It doesn't look related to the device specifically, because it happens with different ones (different number of channels, dimmable and not dimmable), but yes, it seems to happen only  to "light" devices (in my case here).

The most commom thing that happens here is that an automation sends a "turn off" command to the device, on HA it shows up as "off" correctly, but in reality the device is still on.
Not sure if this is a bug related to the "optimistic" functionality.

bernardesarthur, 2025-05-05:
Which firmware are you using? I have a EFR32MG21 Sonoff Dongle E with Ember 7.4.5.0 and I don't have this problems.

pauloon, 2025-05-05:
> Which firmware are you using? I have a EFR32MG21 Sonoff Dongle E with Ember 7.4.5.0 and I don't have this problems.

My coordinator is running on 20250321

kozlowski-mathias, 2025-05-05:
Sonoff Dongle E, EmberZNet 7.4.3 [GA]

bernardesarthur, 2025-05-06:
Any of you are using a GLEDOPTO bulb? Exist a known related bug in that bulbs. Their crash your Zigbee network.

pauloon, 2025-05-06:
> Any of you are using a GLEDOPTO bulb? Exist a known related bug in that bulbs. Their crash your Zigbee network.

No Gledopto devices here.
About my network crashing, it was fixed with a firmware upgrade to the coordinator, as instructed here: https://github.com/Koenkk/zigbee2mqtt/issues/26998