Koenkk/zigbee2mqtt #24488: Ikea Inspelning Plug Wattage Issue

Alexofhiswalnut, 2024-10-26:
### What happened?

Hello,
I purchased two Ikea Inspelning plugs. They are correctly recognized by Zigbee2MQTT, and I updated them to the latest version via ZHA. However, above 1000 watts, the power measurement still appears divided by 10.
Do you know if Zigbee2MQTT will address this bug in a future update?

Best regards,

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.3

### Adapter

Sonoff type E

### Setup

Pi4 haos

kender65, 2024-10-27:
Maybe you ran into the same problem mentionend in [#23961].
Try this [#23961 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2369761682)

Great-Chart, 2024-10-28:
It's due to be updated in next release - 1st Nov
https://github.com/Koenkk/zigbee-herdsman-converters/pull/8060

icheered, 2024-12-01:
This just got fixed in the new release! https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0

closed 2025-01-11 by Alexofhiswalnut