Koenkk/zigbee2mqtt #24501: Tuya ZP01 not sending non-occupied state occasionally

jmijn, 2024-10-27:
### What happened?

I bought a lot of Tuya ZP01 devices and i notice that some of them every once in a while do not send the non-occupied state. The occupied state is always sent perfectly fine, but away is not.

This makes lights staying on for a long time for example, which is a problem. I notice that if it gets into the permanent occupied state, it resets as soon as new motion is detected.

Since it seems to be a device issue, i was wondering if its possible to workaround in z2m. For example to configure somewhere that if this device state is occupied for 30 minutes, which it never should if it works correctly, the state is hard reset to away.

I think i can work around it in Homeassistant by using turning lights off after 30 minutes, and listen to last seen as the next event to turn them on. But it would be better if i can fix it more close to the source.

Also wonder of other users experience the same issue with these devices. Thanks all!

### What did you expect to happen?

Device should report not occupied always after x time

### How to reproduce it (minimal and precise)

It occurs every once in a while, no way to reproduce, you just have to be "lucky".

### Zigbee2MQTT version

Latest

### Adapter firmware version

Latest

### Adapter

Sonoff

### Setup

Z2m and hass in docker

NordJung, 2024-10-28:
I have the same problem:
HA
- Core 2024.10.4 

Z2m
- Zigbee2MQTT-Version 1.40.2

Coordinator:
- SLZB-06M

Motion Sensor:
- Tuya ZP01(https://de.aliexpress.com/item/1005006627166271.html?spm=a2g0o.order_list.order_list_main.17.316e5c5f9HXyM3&gatewayAdapt=glo2deu)

melle1969, 2024-11-03:
I had the same problem. In my case it was because the batteries were too new. After a while it resolved itself.

Core                       2024.10.4

Supervisor              2024.10.3

Operating System   13.2

Frontend                20241002.4

Zigbee2MQTT       1.41.0-1

Koordinator           SLZB-06M

jmijn, 2024-11-04:
Too new? As in, too high voltage? Weird thing is that the issue seems to have been solved at my place too since a few days. I blamed the Zigbee network being self healing and i thought the routes were more optimized after time passed. 🤷‍♂️

melle1969, 2024-11-05:
At least that's how it was with me. The routes are always the same. At first the 'recognized' status was present very often and for a long time. Then I threatened to get a replacement ;) and got the Vallhorn from IKEA. Everything was back to normal the same day.
I used the ZP01 sensor as a sample elsewhere and it worked without any problems from the start. Then I ordered two more and then the problems. You don't know which batch will come from CN.

closed 2025-01-12 by github-actions[bot]