Koenkk/zigbee2mqtt #24512: Firmware is newer than the one available online / Aqara Presence sensor FP1 RTCZCGQ11LM

uncle-fed, 2024-10-27:
### What happened?

I purchased 4 identical Aqare FP1 sensors over a year ago and immediately started using 3 of them.

During the course of the year the firmware update was detected (v58) and all 3 were successfully updated to it (from v54).
Now I connected the 4th sensor which was never updated (with v54 firmware still on it).

### What did you expect to happen?

I expected that I would be able to upgrade it just like the previous three sensors.

### How to reproduce it (minimal and precise)

The following happens when checking for update:

- The v54 sensor gets a response 'No firmware update available'
- The v58 sensors get 'Firmware is newer than the one available online' 

Clearly, I was able to update previously (not sure exactly when, probably very early in 2024 maybe end of 2023) and now the v58 firmware is not available. Has it been pulled away by Aqara or is there some bug that got introduced with z2m during this year?

### Zigbee2MQTT version

1.40.2 commit: e06848d

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Plain Z2M on Pi 4

fuzzy01, 2024-10-31:
See https://github.com/Koenkk/zigbee-OTA/pull/443

uncle-fed, 2024-10-31:
Thanks for the info! Unfortunately, after reading this, I feel like I am in a minority with absolutely zero issues and version 58 since it came out (which is many many months ago). 

I admit that I am not using all the features like regions, but I am relying on the 'presence event' attribute and tha never failed me...

Strange

closed 2024-11-06 by uncle-fed