Koenkk/zigbee2mqtt #24516: [New device support]: TS0601__TZE284_kobbcyum Tongou Zigbee Smart Electrical Circuit Breaker

astulikov24, 2024-10-28:
### Link

https://www.tongou.com/product/zigbee-smart-circuit-breaker-rcbo-tosmr1/

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x94b216fffece0cae","nwkAddr":33135,"manufId":4098,"manufName":"_TZE284_kobbcyum","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�?�.i�?�.i�?�.i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_kobbcyum","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1729765914430}

### Zigbee2MQTT version

1.40.2

### Comments

Suppot for shis device was added (https://github.com/Koenkk/zigbee2mqtt/issues/23054) but my device have another zigbee manufactured - _TZE284_kobbcyum. This manufactured will work if use the same external definition but change manufactured on _TZE284_kobbcyum

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_kobbcyum',
        },
    ],
    
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Single-phase multifunction energy meter (DIN Module)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Important: respondToMcuVersionResponse should be false otherweise there are an avalanche of commandMcuVersionResponse messages every second.
    // queryIntervalSeconds: is doing a pooling to update device's parameters, now define to update data every 3 minutes.
    onEvent: tuya.onEvent({respondToMcuVersionResponse: false, queryIntervalSeconds: 10}),
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch(),
        e.energy(),
        e.power(),
        e.voltage(),
        e.current(),
        e.temperature(),
        e.enum('last_event', ea.STATE, [])
            .withDescription('Last event'),
        e.numeric('leakage_current', ea.STATE).withUnit('mA').withDescription('Leakage current is current that flows from an electrical circuit to the ground'),
        e.enum('leakage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached - USE WITH CAUTION - RISK LIFE !!'),
        e
            .numeric('leakage_threshold', ea.STATE_SET)
            .withValueMin(30)
            .withValueMax(300)
            .withValueStep(1)
            .withUnit('mA')
            .withDescription('leakage threshold setting - USE WITH CAUTION - RISK LIFE !! (default=30mA)'),
        e.enum('over_current_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('current_threshold', ea.STATE_SET)
[... 71 lines cut ...]
                    'Manual Off': tuya.enum(14),
                    'Value 15': tuya.enum(15),
                    'Value 16': tuya.enum(16),
                    'Factory Reset': tuya.enum(17),
                }),
            ],
            [112, 'clear_fault', tuya.valueConverter.onOff],
            [113, 'factory_reset', tuya.valueConverter.raw],
            [114, 'current_threshold', tuya.valueConverter.raw],
            [115, 'over_voltage_threshold', tuya.valueConverter.raw],
            [116, 'under_voltage_threshold', tuya.valueConverter.raw],
            [117, 'leakage_threshold', tuya.valueConverter.raw],
            [118, 'temperature_threshold', tuya.valueConverter.divideBy10],
            [119, 'over_power_threshold', tuya.valueConverter.raw],
            [131, 'temperature', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

There is still some problem: the data is not updated every 10 seconds, the update occurs when the parameters change (for example, switching Auto-Reclosing)

dimacoupe, 2024-10-28:
i think difference in manufacturer because _TZE204_kobbcyum 1-20A, TZE284_kobbcyum 20-40A.
my TOWSMR1 is TZE284_kobbcyum too

terzo33, 2024-11-09:
> i think difference in manufacturer because _TZE204_kobbcyum 1-20A, TZE284_kobbcyum 20-40A.
> my TOWSMR1 is TZE284_kobbcyum too 

My RCBO is TZE204_KOBBCYUM with 20-40A

astulikov24, 2024-11-09:
> > i think difference in manufacturer because _TZE204_kobbcyum 1-20A, TZE284_kobbcyum 20-40A.
> > my TOWSMR1 is TZE284_kobbcyum too
> 
> My RCBO is TZE204_KOBBCYUM with 20-40A

You are lucky :)

Koenkk, 2024-12-02:
Added! Will be supported in the next release.

astulikov24, 2024-12-05:
> Added! Will be supported in the next release.

Many thanks!!

closed 2024-12-05 by Koenkk