Koenkk/zigbee2mqtt #24527: [New device support]: Model TS0601_TZE204_wktrysab 8-gang wall touch switch board

rhakbari, 2024-10-29:
### Link

https://www.aliexpress.com/item/1005005847728198.html?spm=a2g0o.order_list.order_list_main.22.27c11802GJYCK7

### Database entry

{   "FriendlyName": "Touch Switch board",   "Description": "Last seen",   "Availability": "Disabled",   "DeviceType": "Router",   "ZigbeeModel": "TS0601",   "ZigbeeManufacturer": "_TZE204_wktrysab",   "SupportStatus": "Not supported",   "IEEEAddress": "0xa4c138766e42ced8",   "NetworkAddress": "0x9837 / 38967",   "Power": "Interview completed",   "InterviewCompleted": true }

### Zigbee2MQTT version

1.40.2 

### Comments

Hi, 

I paired this 8-gang touch switch board on zigbee2mqtt but it is being shown unsupported in the listing and I am unable to operate the switches by zigbee2mqtt or by home assistant as I am using Tasmota Ewelink wired Zigbee gateway with it. 

Please help me with this issue .

Thanks

### External definition

```shell
// This is the automatically generated external defination

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_wktrysab',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Hi, 

I paired this 8-gang touch switch board on zigbee2mqtt but it is being shown unsupported in the listing and I am unable to operate the switches by zigbee2mqtt or by home assistant as I am using Tasmota Ewelink wired Zigbee gateway with it. 

Please help me with this issue .

Thanks

rhakbari, 2024-11-06:
I have contributed to the repo and added the support for the mentioned device with PR [#8224](https://github.com/Koenkk/zigbee-herdsman-converters/issues/8224)

closed 2024-11-06 by rhakbari

alisian1992-cmyk, 2025-09-22:
hi i bought the same device from the same AliExpress store but a 6 gang, you think this would work for me?

export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_cduqh1l0',
    description: 'Automatically generated 

I'm new to this and I've been trying to make it work for days now, any help would be appreciated.

https://www.aliexpress.com/item/1005005898017809.html?spm=a2g0o.order_list.order_list_main.22.68eb1802ixtSfy
its not advertised as a tuya device btw.