Koenkk/zigbee2mqtt #24528: Detect _TZ3210_3mpwqzuu as TS110E_2gang_2

RoGu777, 2024-10-29:
### What happened?

After installing a _TZ3210_3mpwqzuu I've experienced the same issue as in #23967

Creating an external converter with definitions as of TS110E_2gang_2 solves the problem

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    
       
        fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_pagajpog', '_TZ3210_4ubylghk', '_TZ3210_vfwhhldz', '_TZ3210_3mpwqzuu']),
        model: 'TS110E_2gang_2',
        vendor: 'Tuya',
        description: '2 channel dimmer',
        fromZigbee: [fz.TS110E, fz.TS110E_light_type, tuya.fz.power_on_behavior_1, fz.on_off],
        toZigbee: [tz.TS110E_onoff_brightness, tz.TS110E_options, tuya.tz.power_on_behavior_1, tz.light_brightness_move],
        meta: {multiEndpoint: true},
        exposes: [
            e.light_brightness().withMinBrightness().withMaxBrightness().withEndpoint('l1'),
            e.light_brightness().withMinBrightness().withMaxBrightness().withEndpoint('l2'),
            e.power_on_behavior().withAccess(ea.ALL),
        ],
        configure: async (device, coordinatorEndpoint) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await reporting.onOff(endpoint);
        },
        endpoint: (device) => {
            return {l1: 1, l2: 2};
        },

};

module.exports = definition;

```

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

zbdongle-p

### Setup

Docker

RoGu777, 2024-10-30:
> @RoGu777 Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

done with  #[https://github.com/Koenkk/zigbee-herdsman-converters/pull/8228](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8228), but some checks seem to fail

RoGu777, 2024-10-30:
> @RoGu777 it only seems that the brightness settings like min and max value still do not get applied

on my setup min and max work fine (only for the physical switch of course)

Koenkk, 2024-10-31:
Merged, thanks!

closed 2024-10-31 by Koenkk

kiksia95, 2024-11-17:
I have check that and noticed another issues (not for me) Transition attribute was lost 
[image]

and for some reason I am unable to turn on the light with specific brightness
`target:
  entity_id: light.prawa
data:
  brightness: "{{(states('input_number.maly_pokoj_swiatlo_prawa'))|int(0)}}"
action: light.turn_on
alias: Turn RIGHT ON, recover previous brightness
`
HA frontend and Exposes tab in Z2m show that lights are turned on with provided brightness, but lights are physically not working.

RoGu777, 2024-11-17:
What do you mean with "lost". Was it available before?

I don't see this attribute and can't use it.

kiksia95, 2024-11-18:
> What do you mean with "lost". Was it available before?
> 
> I don't see this attribute and can't use it.

Before I applied the converer the Z2M was not showing any error related with 'transition', so I believe it was working before.

Another thing I've noticed. If we will turn off the light with brightness eg. 100 and then try to turn it on with the same brightness using automation (memory), then there's no physicall reaction, but HA frontend and Z2M exposes tab shows that light is on. 
The automation I used
`target:
  entity_id: light.prawa
action: light.turn_on
alias: Turn RIGHT ON, recover previous brightness
data:
  brightness: 100`

RoGu777, 2024-11-18:
> > What do you mean with "lost". Was it available before?
> > I don't see this attribute and can't use it.
> 
> Before I applied the converer the Z2M was not showing any error related with 'transition', so I believe it was working before.

well, then the new converter version is not 100% fine. Maybe you can revert to the old converter and compare what is the difference in code of the functions.
There are more things I don't like: 
- the brightness value in Z2M is not updated in realtime, only after the physical switch is released
- binding is not working

> Another thing I've noticed. If we will turn off the light with brightness eg. 100 and then try to turn it on with the same brightness using automation (memory), then there's no physicall reaction, [...]

yes, I have that too. As a workaround I execute 2 actions in HA: 1 - turn on the light, 2 - set the brightness