Koenkk/zigbee2mqtt #24534: PJ-MGW1203 - Energy sensor not cumulative

cdemi, 2024-10-29:
### What happened?

The energy sensor on the Tuya PJ-MGW1203 is not cumulative:

[image]
[image]

This causes Home Assistant's energy dashboard to work incorrectly (showing negative values)

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20230507

### Adapter

zzh

### Setup

Home Assistant OS

Bibiraid, 2024-12-08:
Hello, have you found a solution to this problem? I have exactly the same problem with a zigbee clamp module that is recognized as a two clamp module and the data does not accumulate on the general energy as you do thank you.
Good days!

soonic6, 2025-01-15:
problem is still present.

Bibiraid, 2025-01-16:
To accumulate kWh, I've created an integral sensor in home assistant, which allows me to integrate it into energy and monitor consumption correctly.

creew, 2025-02-16:
> To accumulate kWh, I've created an integral sensor in home assistant, which allows me to integrate it into energy and monitor consumption correctly.

can you explain how to do this?

mikeluyten, 2025-02-20:
> > To accumulate kWh, I've created an integral sensor in home assistant, which allows me to integrate it into energy and monitor consumption correctly.
> 
> can you explain how to do this?

Devices & Services -> Helpers -> + Create Helper -> Integral Sensor:
Metric Prefix: k (kilo)
Time Unit: Hours
Input Sensor: Your sensors “Power” reading (the one that reads Watts)
Integration Method: Trapezoidal Rule

After creation, ensure to set the new sensor to be unit of Measurement: kWh

I did this on mine a couple of days ago, results on the first full day are: 
From Power company: 13.44kWh
From Integral Sensor: 13.15kWh

I’m calling it a win!

cdemi, 2025-02-20:
The problem with this approach is that if for some reason the sensor doesn't send data, for example home assistant is down (or some other reason), you will have recorded 0 power for that duration. 

The Energy Sensor keeps a cumulative track of power so that even if the zigbee messages fail, the ones that arrive after will still have the cumulative value

mikeluyten, 2025-02-20:
> The problem with this approach is that if for some reason the sensor doesn't send data, for example home assistant is down (or some other reason), you will have recorded 0 power for that duration. [...]

Very true concern, that would cause issues. Have you tried this sensor via a standard zigbee Tuya hub To see if it works correctly there? The bad data appears to be coming raw from the sensor and not as a result of the z2m software, or z2m is somehow continually triggering a cumulative reset on the device constantly.

cdemi, 2025-02-21:
Unfortunately I don't have a Tuya Hub. But if the sensor itself is actually sending the bad data, then I agree, there's nothing much z2m can do

closed 2025-02-21 by cdemi