Koenkk/zigbee2mqtt #24546: [CK-BL702-AL-01] Bulb turn off when color is set

yan-064, 2024-10-29:
### What happened?

Hi, I'm new on z2m and I paired my first device with no problem, a RGB bulb CK-BL702-AL-01.
My issue is that when I set a color, the bulb turned off. The other features works great, on/off, temperature, brightness... But no color.

Thanks a lot.

### How to reproduce it (minimal and precise)

Go on z2m interface and select bulb device => turn it on => select a color => bulb goes off => the only way to turn it back on is to touch the temperature.
[image]

We can see that selecting a color put a black color on the right side of the color x/y line.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

0x26720700

### Adapter

ConbeeII

### Setup

Rpi4 + Jeedom

### Debug log

info 2024-10-30 00:42:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb40ecfd277ad0000', payload '{"brightness":76,"color":{"hue":240,"saturation":100,"x":0.15,"y":0.0509},"color_mode":"xy","color_power_on_behavior":"previous","color_temp":294,"device":{"applicationVersion":1,"dateCode":"20221117","friendlyName":"0xb40ecfd277ad0000","hardwareVersion":0,"ieeeAddr":"0xb40ecfd277ad0000","manufacturerID":4742,"manufacturerName":"eWeLink","model":"CK-BL702-AL-01","networkAddress":28331,"powerSource":"Unknown","softwareBuildID":"1.2.2","stackVersion":0,"type":"Router","zclVersion":8},"do_not_disturb":false,"last_seen":"2024-10-29T23:42:47.633Z","linkquality":176,"state":"ON"}'

yan-064, 2024-12-29:
Hi, up to stay alive because trouble still alive.

Thanks.

closed 2025-05-09 by github-actions[bot]

gaetanvannes, 2026-08-31:
Found the root cause for the "color set → errors / no color feedback" family of reports on this device.

Zigbee2MQTT (tested on 2.13.0) throws in the logs when setting color_temp:

    TypeError: Cannot read properties of undefined (reading 'x')

After that, the device never reports a valid `color_mode` again, so brightness/color show up as null in Home Assistant even though the bulb itself responds fine.

Cause: the upstream definition for this exact fingerprint (`TS0505B` / `_TZ3210_bfwvfyx1`, model `CK-BL702-AL-01_1`, src/devices/tuya.ts) sets `colorTemp.range: [25, 1000]`. That's far wider than this bulb can actually do. A very similar case was already fixed in #27758 by capping the range to 500 — but that fix was never applied to this specific device entry, which still exposes `value_max: 1000` on the latest release.

Fix (external converter, drop-in replacement for the built-in definition, only changes colorTemp.range from [25,1000] to [153,500] — same bounds already used by a neighboring device definition in the same file):

```js
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';
import * as tuya from 'zigbee-herdsman-converters/lib/tuya';

export default {
    fingerprint: tuya.fingerprint('TS0505B', ['_TZ3210_bfwvfyx1']),
    model: 'CK-BL702-AL-01_1',
    vendor: 'eWeLink',
    description: 'Zigbee 3.0 15W led light bulb E27 RGBCW (colorTemp range fix)',
    toZigbee: [tuya.tz.do_not_disturb],
    extend: [
        m.light({
            colorTemp: {range: [153, 500], startup: true},
            effect: true,
            powerOnBehavior: true,
            moveToLevelWithOnOffDisable: true,
            color: {modes: ['hs', 'xy'], enhancedHue: false},
        }),
    ],
    exposes: [tuya.exposes.doNotDisturb()],
    configure: (device, coordinatorEndpoint) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
    },
};
```

Tested both color_temp and hs color modes with this converter — zero errors, color_mode now reports correctly. Happy to open a PR narrowing the range upstream if that's preferred over an external converter workaround.