Koenkk/zigbee2mqtt #24548: zigbee2mqtt keeps crashing several times, before it keeps working

JeroenTuinstra, 2024-10-30:
### What happened?

Z2M crashes several times in a row (like 10 or 20 times) before it seems to just work again. 
In the logs it crashes for 2 different reasons:

```
z2m: Error: SRSP - SYS - stackTune after 6000ms
```
and
```
z2m: TypeError: Cannot read properties of null (reading'length') 
```
I have a log with the two errors from starting to exiting. 

Attached to the zigbee2mqtt are 2 devices: Tuya MTG275-ZB-RL and Tuya MTG235-ZB-RL (both MCU firmware updated to 1.06 to counteract to much chatter from devices). The MTG235-ZB-RL is curently offline. 

### Zigbee2MQTT version

1.40.2 commit: e06848d

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

Docker

### Debug log

[log-with-error2.log](https://github.com/user-attachments/files/17570318/log-with-error2.log)
[log-with-error1.log](https://github.com/user-attachments/files/17570333/log-with-error1.log)

lux73, 2024-10-30:
https://github.com/Koenkk/zigbee2mqtt/discussions/24364

could be related

JeroenTuinstra, 2024-10-30:
This is the configuration yaml:

```
homeassistant: true
frontend:
  port: 8089
  auth_token: ############
  url: https://##########.org
mqtt:
  base_topic: zigbee2mqtt
  user: ###########
  password: ###########
  server: mqtt://192.168.10.2:1883
serial:
  port: tcp://192.168.10.3:6638
  baudrate: 115200
  adapter: zstack
advanced:
  transmit_power: 20
  network_key:
    - 97
    - 25
    - 153
    - 24
    - 225
    - 160
    - 140
    - 149
    - 211
    - 119
    - 226
    - 137
    - 183
    - 246
    - 67
    - 160
  pan_id: 34120
  ext_pan_id:
    - 55
    - 203
    - 151
    - 96
    - 46
    - 232
    - 143
    - 18
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
device_options:
  legacy: false
devices:
  '0x90395efffe3f3ffb':
    friendly_name: Office Presence Sensor
  '0x0ceff6fffeac3d4e':
    friendly_name: Bedroom Presence Sensor
permit_join: true
```

Zigbee2Mqtt doesn't find an update to 1.41.0. Where would I find that update?

JeroenTuinstra, 2024-10-30:
This is what SLZB-06 recommends to put in the yaml:
[image]

closed 2025-01-07 by github-actions[bot]