Koenkk/zigbee2mqtt #24563: [New device support]: Thermostat radiator valve MZVT1Z

lhelou, 2024-10-31:
### Link

no link, Model Mazda : MZVT1Z

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13825699aa549","nwkAddr":1643,"manufId":4417,"manufName":"_TZE204_k6rdmisz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"�j�.i�o�.i�o�.i\nu�.i\nu�.i4z�.i4z�.in�.i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_k6rdmisz","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1564678876},"lastSeen":1730396160651}

### Zigbee2MQTT version

1.40.2-1

### Comments

Hello,
I tried to follow the supported new device page and i don't succed, could you help me please ?
Thanks
Lionel

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_k6rdmisz',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The device is not reconized

hopgun, 2024-11-15:
Hello. I have the same model and I'm not able to add the new device as well. 

I tried the following js file as extension but nothing happen:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['MZVT1Z'],
    model: 'MZVT1Z',
    vendor: 'Mazda',  // Mets le nom du fabricant si tu le connais
    description: 'Thermostat pour radiateur',
    fromZigbee: [fz.thermostat, fz.battery],  // Ajuste ces valeurs si besoin
    toZigbee: [tz.thermostat_system_mode, tz.thermostat_occupied_heating_setpoint],
    extend: [],
    exposes: [
        e.battery(),
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)  // Plage de température (5 à 30°C)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat', 'auto'])  // Modes disponibles
            .withRunningState(['idle', 'heat']),
    ],
};

module.exports = definition;
`

Is anyone have a detailed guideline ? I have 10 of those thermostats :)

Doable149, 2024-11-30:
Hello!

So I tried some things but did not succeed to make the schedule part work correctly, each time I try to put my thermostat in planning mode it makes the wanted temperature go to 1674,1°C. 

Here is my config for now : 

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_k6rdmisz',
        },
    ],
    model: 'TS0601_new',
    vendor: 'Tuya',
    description: 'Thermostatic radiator valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        e.child_lock(),
        e.temperature(),
        e.window_detection(),
        tuya.exposes.frostProtection(),
        e.binary('window', ea.STATE, 'CLOSED', 'OPEN').withDescription('Window status closed or open '),
        e.binary('alarm_switch', ea.STATE, 'ON', 'OFF').withDescription('Thermostat in error state'),
        e.comfort_temperature().withValueMin(5).withValueMax(35).withDescription('Comfort mode temperature'),
        e.eco_temperature().withValueMin(5).withValueMax(35).withDescription('Eco mode temperature'),
        e.holiday_temperature().withValueMin(5).withValueMax(35).withDescription('Holiday mode temperature'),
        e
                .numeric('temperature_sensitivity', ea.STATE_SET)
                .withUnit('°C')
                .withValueMin(0.5)
                .withValueMax(5)
                .withValueStep(0.5)
                .withDescription('Temperature sensitivity'),
        e
                .text('schedule_monday_raw', ea.STATE_SET)
                .withDescription('Monday schedule raw'),
        e.enum('mode', ea.STATE_SET, ['standby', 'auto', 'eco', 'comfort', 'frost_protection', 'on']).withDescription('Mode'),
        e.climate()
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withSystemMode(['off', 'heat'], ea.STATE_SET, 'Only for Homeassistant')
            .withLocalTemperatureCalibration(-9.5, 9.5, 0.5, ea.STATE_SET),

        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints: [
            [2,
                'mode',
                tuya.valueConverterBasic.lookup({
                    standby: tuya.enum(0),
                    auto: tuya.enum(1),
                    eco: tuya.enum(2),
                    comfort: tuya.enum(3),
                    frost_protection: tuya.enum(4),
                    on: tuya.enum(5),
                })
            ],
            [3, 'running_state', tuya.valueConverterBasic.lookup({ 'heat': 1, 'idle': 0 })],
            [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [5, 'local_temperature', tuya.valueConverter.divideBy10],
            [6, 'battery', tuya.valueConverter.raw],
            [7, 'child_lock', tuya.valueConverter.lockUnlock],
            [103, 'eco_temperature', tuya.valueConverter.divideBy10],
            [104, 'comfort_temperature', tuya.valueConverter.divideBy10],
            [105, 'frost_temperature', tuya.valueConverter.divideBy10],
            [102, 'temperature_sensitivity', tuya.valueConverter.divideBy10],
            [21, 'holiday_temperature', tuya.valueConverter.divideBy10],
            [15, 'window', tuya.valueConverterBasic.lookup({ 'OPEN': 1, 'CLOSE': 0 })],
            [14, 'window_detection', tuya.valueConverter.onOff],
            [35, 'alarm_switch', tuya.valueConverter.onOff],
            [36, 'frost_protection', tuya.valueConverter.onOff],
            [28, 'schedule_monday_raw', tuya.valueConverter.raw],
            [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(1)],
            [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(2)],
            [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(3)],
            [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(4)],
            [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(5)],
            [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(6)],
            [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP_TRV602Z_WithDayNumber(7)],
            [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
        ],
    },
};

module.exports = definition;
```

lhelou, 2024-12-04:
Hello,
So with this configuration file you can operate the thermostatic valve.
I tried and it doesn't work for me.
I have a new file (TS0601.js) included from the configuration.yaml file as external converter and yet nothing appears in the data exposed in the Zigbee2mqtt administration page.
Do you have any idea ?

Thanks
Lionel

[image]
[image]

lhelou, 2024-12-04:
I have found the problem,
in your file you have "_TZE284_k6rdmisz" and my sensor is "_TZE204_k6rdmisz"
lets go for the tests

Nopraz, 2024-12-06:
Yes sorry about that, I found out later about the differences in models. The pull requests includes your model

lhelou, 2024-12-06:
Hello, 
Thanks for your work.
I have done my tests, some values are ok (battery, current heating set point, local temperature) but other not.

[image]
[image]
[image]

Do you have any idea ?

Thanks.
Lionel

Nopraz, 2024-12-06:
Since the pull request I modified the javascript file, here is the new one : 

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

tuya.valueConverter.myCustomConverter = {
    from: (v) => {
        const schedule = [];
        for (let index = 1; index < 24; index = index + 4) {
            schedule.push(
                String(parseInt(v[index + 0])).padStart(2, '0') +
                    ':' +
                    String(parseInt(v[index + 1])).padStart(2, '0') +
                    '/' +
                    (parseFloat((v[index + 2] << 8) + v[index + 3]) / 10.0).toFixed(1),
            );
        }
        return schedule.join(' ');
    },
    to: (v) => {
        const payload = [];
        const transitions = v.split(' ');
        if (transitions.length != 6) {
            throw new Error('Invalid schedule: there should be 6 transitions');
        }
        for (const transition of transitions) {
            const timeTemp = transition.split('/');
            if (timeTemp.length != 2) {
                throw new Error('Invalid schedule: wrong transition format: ' + transition);
            }
            const hourMin = timeTemp[0].split(':');
            const hour = parseInt(hourMin[0]);
            const min = parseInt(hourMin[1]);
            const temperature = Math.floor(parseFloat(timeTemp[1]) * 10);
            if (hour < 0 || hour > 24 || min < 0 || min > 60 || temperature < 50 || temperature > 300) {
                throw new Error('Invalid hour, minute or temperature of: ' + transition);
            }

            const firstByte = hour;
            const secondByte = min;

            payload.push(firstByte, secondByte, 0, temperature);
        }
        return payload;
    },
};

tuya.valueConverter.myCustomConverterWithDayNumber = (dayNum) => {
    return {
        from: (v) => tuya.valueConverter.myCustomConverter.from(v),
        to: (v) => {
            const data = tuya.valueConverter.myCustomConverter.to(v);
            data.unshift(dayNum);
            return data;
        },
    }
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_k6rdmisz',
        },
    ],
    model: 'MZVT1Z',
    vendor: 'MAZDA',
[... 57 lines cut ...]
            [6, 'battery', tuya.valueConverter.raw],
            [7, 'child_lock', tuya.valueConverter.lockUnlock],
            [103, 'eco_temperature', tuya.valueConverter.divideBy10],
            [104, 'comfort_temperature', tuya.valueConverter.divideBy10],
            [105, 'frost_temperature', tuya.valueConverter.divideBy10],
            [102, 'temperature_sensitivity', tuya.valueConverter.divideBy10],
            [21, 'holiday_temperature', tuya.valueConverter.divideBy10],
            [15, 'window', tuya.valueConverterBasic.lookup({ 'OPEN': 1, 'CLOSE': 0 })],
            [14, 'window_detection', tuya.valueConverter.onOff],
            [35, 'alarm_switch', tuya.valueConverter.onOff],
            [36, 'frost_protection', tuya.valueConverter.onOff],
            [28, 'schedule_monday', tuya.valueConverter.myCustomConverterWithDayNumber(1)],
            [29, 'schedule_tuesday', tuya.valueConverter.myCustomConverterWithDayNumber(2)],
            [30, 'schedule_wednesday', tuya.valueConverter.myCustomConverterWithDayNumber(3)],
            [31, 'schedule_thursday', tuya.valueConverter.myCustomConverterWithDayNumber(4)],
            [32, 'schedule_friday', tuya.valueConverter.myCustomConverterWithDayNumber(5)],
            [33, 'schedule_saturday', tuya.valueConverter.myCustomConverterWithDayNumber(6)],
            [34, 'schedule_sunday', tuya.valueConverter.myCustomConverterWithDayNumber(7)],
            [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
        ],
    },
};

module.exports = definition;
```

hopgun, 2024-12-15:
Hello,

I'm testing this setup with zigbee2mqtt and node red and it seems working :-)

_(Initially, I didn't receive all fields from mqtt, but after I get it but I don't how)_

After some tests, I have some issues.

First, the /get topic doesn't work or I didn't use it correctly.

Second, to have an update of the status, I send the following message to the topic /set but I always receive the message twice with the request:

`msg.payload =
{
 child_lock: "LOCK",
};`

Recived message:

`object
battery: 69
child_lock: "LOCK"
comfort_temperature: 21
current_heating_setpoint: 23
eco_temperature: 15
holiday_temperature: 10
linkquality: 208
local_temperature: 21.5
local_temperature_calibration: 0
preset: "manual"
running_state: "idle"
temperature_sensitivity: 3
window_detection: "OFF"`

To setup the valve, I send the following message but I received 18 messages as feedback !

`msg.payload =
{
    preset: "manual",//manual comfort eco schedule
    child_lock: "LOCK",//UNLOCK LOCK
    window_detection: "OFF",//OFF ON
    current_heating_setpoint: 23,
    comfort_temperature: 21,
    eco_temperature: 15,
    holiday_temperature: 10,
    local_temperature_calibration: 0,
    temperature_sensitivity: 3,
};`

It's working but not correctly (at least in node red)...

Jean-Philipp, 2025-01-03:
Hello,
First of all, happy new year to everyone!
I just go 9 thermosatic radiator valves from MAZDA. I set up the whole Home Assistant with Zigbee2mqtt, to be able to control automatically the valves and get a more user friendly interface, than their native app. 
I was disappointed to see, that these valves are not supported yet by Zigbee2Mqtt...
@Nopraz, your Javascript file helped a lot, but it is still not working properly. 
Are there any news on this topic, since many person got these valves and are facing similar issues? 

Thanks a lot in advance. 
I'm new GitHub and to Home Assistant, so if I missed something, don't hesitate to let me know :)
Thanks a lot!

closed 2025-03-05 by lhelou