Koenkk/zigbee2mqtt #24565: Wont start after migrate to new coorniator

ChillZwix, 2024-10-31:
### What happened?

Did change from Sonoff to SLZB-06M.

Did follow https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another and deleted data/coordinator_backup.json  and data/database.db

Now zigbee2mqtt just crashes and shuts down.

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

 20240510

### Adapter

SLZB-06M

### Setup

HAOS on Generic x86-64

### Debug log

My error from the log:

[19:40:45] INFO: Preparing to start...
[19:40:45] INFO: Socat not enabled
[19:40:46] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
RangeError: The value of "value" is out of range. It must be >= 0 and <= 65535. Received 3792000638
    at new NodeError (node:internal/errors:405:5)
    at checkInt (node:internal/buffer:74:11)
    at writeU_Int16LE (node:internal/buffer:724:3)
    at Buffer.writeUInt16LE (node:internal/buffer:732:10)
    at BuffaloZdo.writeUInt16 (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:40:21)
    at BuffaloZdo.buildIeeeAddressRequest (/app/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:906:14)
    at Function.buildRequest (/app/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:767:32)
    at Controller.identifyUnknownDevice (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:542:40)
    at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:850:37)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
[19:41:09] INFO: Preparing to start...
[19:41:09] INFO: Socat not enabled
[19:41:10] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[31-10-2024 19:41:19] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[31-10-2024 19:41:19] error: 	zh:ezsp:uart: -!- break waiting (1)
[31-10-2024 19:41:19] error: 	zh:ezsp:uart: Can't send DATA frame (0,2,0): 00000004
[31-10-2024 19:41:23] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[31-10-2024 19:41:23] error: 	zh:ezsp:uart: -!- break rewaiting (1)
[31-10-2024 19:41:23] error: 	zh:ezsp:uart: Can't resend DATA frame (0,2,1): 00000004
[31-10-2024 19:41:23] error: 	z2m: Error while starting zigbee-herdsman
[31-10-2024 19:41:23] error: 	z2m: Failed to start zigbee
[31-10-2024 19:41:23] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[31-10-2024 19:41:23] error: 	z2m: Exiting...
[31-10-2024 19:41:23] error: 	z2m: Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:679:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:666:16)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:457:24)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:195:9)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:152:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)

ChillZwix, 2024-10-31:
Found the problem, needed to change adapter to "ember", now it works.

But still one problem:

[22:53:56] INFO: Preparing to start...
[22:53:57] INFO: Socat not enabled
[22:53:57] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[31-10-2024 22:53:59] error: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.

Any one?

ChillZwix, 2024-11-01:
This is so damn unstable.. My smart home is down...

[19:02:41] INFO: Preparing to start...
[19:02:41] INFO: Socat not enabled
[19:02:42] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[01-11-2024 19:02:47] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=RESET_ASSERT.
[01-11-2024 19:02:47] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[01-11-2024 19:02:47] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[01-11-2024 19:02:47] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[01-11-2024 19:02:47] error: 	z2m: Adapter disconnected, stopping
[01-11-2024 19:02:50] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:114:18)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.stop (/app/lib/controller.ts:265:9)
    at Controller.onZigbeeAdapterDisconnected (/app/lib/controller.ts:292:9)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[19:02:51] INFO: Preparing to start...
[19:02:51] INFO: Socat not enabled
[19:02:51] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[01-11-2024 19:02:55] error: 	z2m: Failed to set permit join to false (~x~> [ZDO PERMIT_JOINING_REQUEST BROADCAST to=65532 messageTag=1] Failed to send request with status=BUSY.)
[01-11-2024 19:02:57] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=RESET_ASSERT.
[01-11-2024 19:02:57] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[01-11-2024 19:02:57] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[01-11-2024 19:02:57] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[01-11-2024 19:02:57] error: 	z2m: Adapter disconnected, stopping
[01-11-2024 19:03:00] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:114:18)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.stop (/app/lib/controller.ts:265:9)
    at Controller.onZigbeeAdapterDisconnected (/app/lib/controller.ts:292:9)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[19:17:48] INFO: Preparing to start...
[19:17:48] INFO: Socat not enabled
[19:17:48] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[01-11-2024 19:18:44] error: 	z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: '~x~> [ZDO PERMIT_JOINING_REQUEST BROADCAST to=65532 messageTag=4] Failed to send request with status=BUSY.'

closed 2025-01-09 by github-actions[bot]

kthofmann, 2025-03-10:
Did you found any solution for your issue?