Koenkk/zigbee2mqtt #24568: Node JS 23 Incompatibility

JP-Ellis, 2024-11-01:
### What happened?

The frontend fails to load on NodeJS 23, resulting in the whole process crashing

### What did you expect to happen?

For the front end to start

### How to reproduce it (minimal and precise)

Run Zigbee2MQTT on Node.JS, with front end enabled.

### Zigbee2MQTT version

1.40.2-1 (Arch Linux)

### Adapter firmware version

N/A

### Adapter

N/A

### Setup

Arch Linux

### Debug log

```
Nov 01 15:16:27 asrock node[428700]: [2024-11-01 15:16:27] info:         z2m: Started frontend on port 0.0.0.0:47289
Nov 01 15:16:27 asrock node[428700]: /usr/share/webapps/zigbee2mqtt/node_modules/@folder/readdir/lib/async.js:256
Nov 01 15:16:27 asrock node[428700]:   return new Promise((resolve, reject) => {
Nov 01 15:16:27 asrock node[428700]:          ^
Nov 01 15:16:27 asrock node[428700]: TypeError: Cannot set property path of #<Dirent> which has only a getter
Nov 01 15:16:27 asrock node[428700]:     at /usr/share/webapps/zigbee2mqtt/node_modules/@folder/readdir/lib/async.js:260:17
Nov 01 15:16:27 asrock node[428700]:     at new Promise (<anonymous>)
Nov 01 15:16:27 asrock node[428700]:     at readdir (/usr/share/webapps/zigbee2mqtt/node_modules/@folder/readdir/lib/async.js:256:10)
Nov 01 15:16:27 asrock node[428700]:     at createCache (/usr/share/webapps/zigbee2mqtt/node_modules/connect-gzip-static/lib/gzip-static.js:19:23)
Nov 01 15:16:27 asrock node[428700]:     at createMethods (/usr/share/webapps/zigbee2mqtt/node_modules/connect-gzip-static/lib/gzip-static.js:32:54)
Nov 01 15:16:27 asrock node[428700]:     at module.exports (/usr/share/webapps/zigbee2mqtt/node_modules/connect-gzip-static/lib/gzip-static.js:38:26)
Nov 01 15:16:27 asrock node[428700]:     at Frontend.start (/usr/share/webapps/zigbee2mqtt/lib/extension/frontend.ts:89:37)
Nov 01 15:16:27 asrock node[428700]:     at Controller.callExtensions (/usr/share/webapps/zigbee2mqtt/lib/controller.ts:399:42)
Nov 01 15:16:27 asrock node[428700]:     at processTicksAndRejections (node:internal/process/task_queues:105:5)
Nov 01 15:16:27 asrock node[428700]:     at Controller.start (/usr/share/webapps/zigbee2mqtt/lib/controller.ts:218:9)
Nov 01 15:16:27 asrock systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 15:16:27 asrock systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

bdolgov, 2024-11-03:
Indeed, `Dirent.path` became read-only in node 23: https://nodejs.org/api/fs.html#direntpath.

But `Dirent.path` is heavily used by https://github.com/folder/readdir, which was a dependency of `connect-gzip-static`. Since https://github.com/pirxpilot/connect-gzip-static/commit/f227d83c762e8d026277235dce92b6130cd314b2, `connect-gzip-static` doesn't depend on `@folder/readdir` anymore. So upgrading `connect-gzip-static` to `3.0.1` or `4.0.0` should fix the problem.

JP-Ellis, 2024-11-04:
Thanks for the quick fix! Looking forward to the next release :)

JP-Ellis, 2025-01-04:
Will confirm whether this is fixed in the recently released `2.0.0` when I upgrade (probably not this weekend, but hopefully next weekend).

Koenkk, 2025-02-07:
connect-gzip-static has been replaced with express-static-gzip so I assume this is fixed now.

PR #24619 (2024-11-04): Replace connect-gzip-static with express-static-gzip to become compatible with node 23.
Removes a dependency on `@folder/readdir` which is not compatible with node 23.

Fixes #24568.
closed 2025-02-07 by Koenkk