Koenkk/zigbee2mqtt #24571: [New device support]: Nedis ZBSWM20WT PIR-sensor (_TZE200_s6hzw8g2)

jovandeginste, 2024-11-01:
### Link

https://nedis.com/en-us/product/safety-and-security/detectors/motion-detector/550726145/smartlife-motion-sensor-zigbee-30-battery-powered-ip20-detector-angle-120-0-detector-range-5-m-max-battery-life-12-months-white

### Database entry

{"id":61,"type":"EndDevice","ieeeAddr":"0xa4c1383d93f21bcd","nwkAddr":34925,"manufId":4417,"manufName":"_TZE200_s6hzw8g2","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE200_s6hzw8g2","zclVersion":3,"appVersion":67,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1730451834190}

### Zigbee2MQTT version

1.40.2

### Comments

It was not supported out of the box, but got it working after adding this converter: https://github.com/jatail/z2m-nedis-zbswm20wt-converter
(I had to put the `extend` line in comment)

### External definition

```shell
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
// const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
  // Since a lot of TuYa devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  // fingerprint: [
  //     {
  //         // The model ID from: Device with modelID 'TS0601' is not supported
  //         // You may need to add \u0000 at the end of the name in some cases
  //         modelID: 'TS0601',
  //         // The manufacturer name from: Device with modelID 'TS0601' is not supported.
  //         manufacturerName: '_TZE200_s6hzw8g2',
  //     },
  // ],
  fingerprint: tuya.fingerprint("TS0601", ["_TZE200_s6hzw8g2"]),
  model: "ZBSM20WT",
  vendor: "Nedis",
  description: "Nedis motion sensor",
  fromZigbee: [tuya.fz.datapoints],
  //fromZigbee: [fz.ias_occupancy_alarm_1, fz.ias_occupancy_alarm_1_report, fz.battery, fz.illuminance, fz.illuminance_lux],
  toZigbee: [tuya.tz.datapoints],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: tuya.configureMagicPacket,
  exposes: [
    // Here you should put all functionality that your device exposes
    e.occupancy(),
    e.illuminance().withUnit("lx"),
    e.battery(),
    e
      .enum("sensitivity", ea.STATE_SET, ["low", "medium", "high"])
      .withDescription(
        "PIR sensor sensitivity (refresh and update only while active)",
      ),
    e
      .enum("keep_time", ea.STATE_SET, ["10", "30", "60", "120"])
      .withDescription(
        "PIR keep time in seconds (refresh and update only while active)",
      ),
    e
      .numeric("illuminance_interval", ea.STATE_SET)
      .withValueMin(1)
      .withValueMax(720)
      .withValueStep(1)
      .withUnit("minutes")
      .withDescription(
        "Brightness acquisition interval (refresh and update only while active)",
      ),
  ],
  meta: {
    // All datapoints go in here
    tuyaDatapoints: [
      [1, "occupancy", tuya.valueConverter.trueFalse0],
      [4, "battery", tuya.valueConverter.raw],
      [
        9,
        "sensitivity",
        tuya.valueConverterBasic.lookup({
          low: tuya.enum(0),
          medium: tuya.enum(1),
          high: tuya.enum(2),
        }),
      ],
      [
        10,
        "keep_time",
        tuya.valueConverterBasic.lookup({
          10: tuya.enum(0),
          30: tuya.enum(1),
          60: tuya.enum(2),
          120: tuya.enum(3),
        }),
      ],
      [101, "illuminance", tuya.valueConverter.raw],
      [102, "illuminance_interval", tuya.valueConverter.raw],
    ],
  },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Working

- Motion detecting
- Battery level

Not working

- Illuminance

Not sure if working or not

- Sensitivity setting
- Keep time setting

visdmin, 2024-12-09:
I happen to also have these devices and would love to see support ! <3

jovandeginste, 2025-03-07:
Not stale

jovandeginste, 2025-03-07:
Not stale

closed 2025-05-15 by github-actions[bot]