Koenkk/zigbee2mqtt #24572: OTA not showing for IKEA E2206 (INSPELNING) after upgrade to Z2M 1.41.0

lbschenkel, 2024-11-01:
### What happened?

I have a number of IKEA E2206 (INSPELNING) plugs. They are still on factory firmware, version 2.4.34 and not updated to the latest firmware. Instead of doing workarounds to update them, I was waiting for Z2M 1.41.0 to be released because I knew that it would come with fixes the OTA URL.

However, after upgrading to 1.41.0 and clicking on "check for new updates" for those devices, I still get "no updates available".

### What did you expect to happen?

Given that [#3067](https://github.com/Koenkk/zigbee2mqtt.io/pull/3067) was fixed and incorporated into this release, I was expecting the OTA to work now and for the new firmware to be found.

I also removed and rejoined the device, just in case, but it didn't change the outcome.

### How to reproduce it (minimal and precise)

- Update to 1.41.0
- Go to OTA page, click on "check for new updates"
- Popup with "no update available" is shown

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

ZBDongle-P

### Setup

Home Assistant add-on

### Debug log

```
info 2024-11-01 12:18:14z2m: Checking if update available for 'REDACTED'
info 2024-11-01 12:18:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/REDACTED', payload '{"current":0,"energy":0,"last_seen":"2024-11-01T12:18:16+01:00","linkquality":127,"power":0,"power_on_behavior":"on","state":"ON","update":{"installed_version":33816628,"latest_version":33816628,"state":"idle"},"update_available":null,"voltage":23.2}'
info 2024-11-01 12:18:16z2m: No update available for 'REDACTED'
```

jamesonuk, 2024-11-01:
As per
https://github.com/Koenkk/zigbee2mqtt/discussions/24464
and
https://github.com/Koenkk/zigbee-OTA/commit/35afc409fee009079ddf1f67a1163130084eb1cc

https://github.com/Koenkk/zigbee-herdsman-converters/blob/f54114a3de95b9380f895007a68809ea1a958346/src/devices/ikea.ts#L606C113-L607C18

need to be changed to use zigbeeOTA (probably needs import too)
to get the OTA you are going to either update the definition or create an external converter as in https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2451784804 until someone puts in a PR to change the converter

randommen96, 2024-11-01:
Great workaround for the time being, thanks for the clear steps!

jamesonuk, 2024-11-01:
Created https://github.com/Koenkk/zigbee-herdsman-converters/pull/8248 to add this

treetip, 2024-11-02:
You can just use this custom converter to update all your plugs, and then just remove it (it's built from the current master version, just uses ota instead of ikeaOta)

```
const constants_1 = require("zigbee-herdsman-converters/lib/constants");
const ikea_1 = require("zigbee-herdsman-converters/lib/ikea");
const modernExtend_1 = require("zigbee-herdsman-converters/lib/modernExtend");

const definition = 
    {
        zigbeeModel: ['INSPELNING Smart plug'],
        model: 'E2206',
        vendor: 'IKEA',
        description: 'INSPELNING smart plug',
        extend: [(0, ikea_1.addCustomClusterManuSpecificIkeaUnknown)(), (0, modernExtend_1.onOff)(), (0, modernExtend_1.identify)(), (0, modernExtend_1.ota)(), (0, modernExtend_1.electricityMeter)()],
        configure: async (device) => {
            const endpoint = device.getEndpoint(1);
            // Enable reporting of powerDivisor, needs to change dynamically with the amount of power
            // For details, see: https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2366733453
            await endpoint.configureReporting('haElectricalMeasurement', [
                { attribute: 'acPowerDivisor', minimumReportInterval: 10, maximumReportInterval: constants_1.repInterval.MAX, reportableChange: 1 },
            ]);
        },
    };

module.exports = definition;
```

Koenkk, 2024-11-02:
Fixed now, thanks @jamesonuk !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-02 by Koenkk

rolfie23, 2024-11-03:
> Fixed now, thanks @jamesonuk !
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Updated just few minutes ago to 1.41.0-dev commit: 848f250d when came available but still no update message. In the commit I doesn't see it mentioned as fix.

Koenkk, 2024-11-03:
Were some issues with the update, should be ready once https://github.com/Koenkk/zigbee2mqtt/pull/24609 is merged

rolfie23, 2024-11-03:
> Got some issues with the update, should be ready once #24609 is merged

Okay, don't know where to see if it's merged so can you let me know?

Misiu, 2024-11-07:
> * docker exec -it addon_45df7312_zigbee2mqtt bash

can this be done from Terminal & SSH addon?
when I try to exec that command I get: command not found

[image]

mitch66692, 2024-11-07:
> > Got some issues with the update, should be ready once #24609 is merged
> 
> Okay, don't know where to see if it's merged so can you let me know!
> 
> EDIT: Saw there was an new dev, updated and did get an error: `Failed to check if update available for 'Inspelning stopcontact' (ENOENT: no such file or directory, open '/opt/zigbee2mqtt/data/index.json')`. After putting "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/refs/heads/master/index.json" in OTA index override it was working! it's now updating!

Is it really so simple to fix it? Does OTA index override file name can have any sort of problem if we copy and paste that link?

rolfie23, 2024-11-07:
> Is it really so simple to fix it? Does OTA index override file name can have any sort of problem if we copy and paste that link?

It was after the dev update. Stable update next month its in, after that update did get the error message. This was the solution. Its just an file for an repository wich get updated by koenkk.

Here everything is now updated, besides of the round 5 button remote of IKEA. Gets at 100% and then says it's updated with same firmware. That I'll leave for now, have some other things to do.

adamk125, 2025-01-09:
> Got some issues with the update, should be ready once #24609 is merged

Hi Guys,

I am still having issues updating my Inspelning Smart Plugs even with the latest version of Z2M (not the dev branch). I hoped that the updates were merged into the latest version.

Do I need to do the external converter trick for this to work?

Thanks

[image]

[image]

adamk125, 2025-01-16:
> Hi Guys,
>
> I am still having issues updating my Inspelning Smart Plugs even with the latest version of Z2M (not the dev branch). I hoped that the updates were merged into the latest version.
>
> Do I [...]

Issue resolved finally. See comment in this thread: https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2593997740