Koenkk/zigbee2mqtt #24576: zigbee2mqtt starts every 2 min.

miwok72, 2024-11-01:
### What happened?

After installing 1.41.0 zigbee2mqtt stop and start every 2 min.
With 1.40.2 everything was O.K.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

Slaesht's

### Setup

Pi

### Debug log

[log.log](https://github.com/user-attachments/files/17599888/log.log)

ShafiqUrRehman, 2024-11-03:
I was having the same problem for last 3 day and fixed it few hours back. 
Run

npm install sd-notify

and you should be good.

miwok72, 2024-11-03:
unfortunataly the same behaviour after that ...

nd@ser:~ $ sudo systemctl start zigbee2mqtt

Job for zigbee2mqtt.service failed because a timeout was exceeded.
See "systemctl status zigbee2mqtt.service" and "journalctl -xeu zigbee2mqtt.service" for details.

nd@ser:~ $

and then, zigbee2mqtt restarts every 2 min.

ShafiqUrRehman, 2024-11-03:
try running "npm ls" in "/opt/zigbee2mqtt" folder and check if there are any unmet dependencies are there.

miwok72, 2024-11-03:
same behaviour...

nd@ser:/opt/zigbee2mqtt $ npm ci
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported

added 650 packages, and audited 651 packages in 32s

90 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
nd@ser:/opt/zigbee2mqtt $ npm run build

> zigbee2mqtt@1.41.0 build
> tsc && node index.js writehash

nd@ser:/opt/zigbee2mqtt $

debug log:

[2024-11-03 10:52:04] debug:    zh:zstack:unpi:parser: --- parseNext [254,4,97,50,0,0,0,0,87]
[2024-11-03 10:52:04] debug:    zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [0,0,0,0] - 87
[2024-11-03 10:52:04] debug:    zh:zstack:znp: <-- SRSP: SYS - nvLength - {"len":0}
[2024-11-03 10:52:04] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-11-03 10:52:04] debug:    zh:zstack:backup: fetched adapter network security material table (capacity=2, used=2)
[2024-11-03 10:52:04] debug:    zh:zstack:backup: Following devices with link key are missing from new backup but present in old backup and database, adding them back:                                                               [2024-11-03 10:52:04] info:     zh:controller: Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json'
[2024-11-03 10:52:04] info:     zh:zstack:znp: closing
[2024-11-03 10:52:04] info:     zh:zstack:znp: Port closed
[2024-11-03 10:52:04] info:     z2m: Stopped zigbee-herdsman
[2024-11-03 10:52:04] info:     z2m: Stopped Zigbee2MQTT

ShafiqUrRehman, 2024-11-03:
i am lost now.

stop the zigbee service and then try deleting coordinator_backup.json and database.db.backup in opt/zigbee2mqtt/data folder.

and then restart.

miwok72, 2024-11-03:
many thanks for the help, but unfortunately no success. in the meantime i have also tried the previous versino 1.40.2. this worked normally before. this version now also shows the same behavior and restarts every 2 minutes ...

miwok72, 2024-11-03:
if i start zigbee2mqtt with the command

`cd /opt/zigbee2mqtt`
`npm start  `

there are no problems!
Is there probably a problem with the OS?

```
nd@ser:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
nd@ser:~ $

```

miwok72, 2024-11-03:
see also #22164 and #21463 and #23084

workaround:
https://github.com/Koenkk/zigbee2mqtt/issues/22164#issuecomment-2061199685

closed 2025-01-10 by github-actions[bot]

jerryhjones, 2025-01-16:
I can't say with 100% certainty that this will work for everyone (because i was grasping at straws like so many others), but here's what seemed to work for me.

I was seeing the same things described  by many - constant restarts if started with systemctl, ran fine running from the CLI.

I installed `libsystemd-dev` as indicated [here](https://github.com/Koenkk/zigbee2mqtt/issues/21463#issuecomment-1953157082).

Then I ran this, [as mentioned by ShafiqUrRehman](https://github.com/Koenkk/zigbee2mqtt/issues/24576#issuecomment-2453362676):
```
cd /opt/zigbee2mqtt
npm ci
npm run build
reboot
```

I'm now running swimmingly.