Koenkk/zigbee2mqtt #24578: Nous A4Z does not report power consumption correctly

theincogtion, 2024-11-01:
### What happened?

Watt value is 0 or incorrect (too low)

### What did you expect to happen?

Watt value should be Voltage * Current
in this case
0,03A * 234V = 7Watt

### How to reproduce it (minimal and precise)

Switch on the relay and draw power

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.2 GA

### Adapter

EZSP EmberZNet

### Setup

Addon für HA

### Debug log

[image]
[log.log](https://github.com/user-attachments/files/17599993/log.log)

theincogtion, 2025-01-01:
Issue still present

avilleret, 2025-02-16:
Issue still present.
Any hint where to look to fix that ?

AlexanderLeporiday, 2025-02-20:
This device, according to the converter settings, is polled to get the current power. This polling happens, by default, every 60 seconds but can be configured under settings.
If set to -1 it will never poll.

If the device is not updating with small changes it may be the reporting settings, by default these are set to 10 watts.
Check the reporting page and lower the minimum value.
[image]

closed 2025-04-29 by github-actions[bot]