Koenkk/zigbee2mqtt #24579: Nous A4Z: Firmware Version an OTA Update

theincogtion, 2024-11-01:
### What happened?

Firmware version is not displayed for Nous A4Z

### What did you expect to happen?

FW version should be visible

### How to reproduce it (minimal and precise)

Go to OTA Menu in z2m

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

7.4.2 GA

### Adapter

EZSP EmberZNet

### Setup

Addon for HA

### Debug log

[log.log](https://github.com/user-attachments/files/17600012/log.log)
[image]

Ricc68, 2024-11-02:
This issue is the same that I have reported in #24051 related to the Nous A1Z smart plug device.

closed 2025-01-09 by github-actions[bot]

Ricc68, 2025-05-03:
Fixed by mentioned PR.