Koenkk/zigbee2mqtt #24581: ThirdReality 3RVS01031Z vibration sensor not reporting Accelerometer values

dwaned, 2024-11-01:
### What happened?

I purchased a new ThirdReality 3RVS01031Z vibration sensor, installed it as usual, and noticed that it is not reporting any values for accelerometer. The values are showing as null:
X axis:N/A
Y axis:N/A
Z axis:N/A

Tested in both latest version (1.41.0-1) and previous version.

### What did you expect to happen?

The device should report these values as mentioned here https://www.zigbee2mqtt.io/devices/3RVS01031Z.html

### How to reproduce it (minimal and precise)

There is no steps to reproduce, just install and observe in Zigbee2MQTT device screen.

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet

### Setup

Pi

RobLeighton67, 2024-11-04:
I have the same T.R. device running fine on a Sonoff P adapter - sensor sw version is 40 , an OTA is offered to version 53 which I haven't performed.
What version is your Vibration Sensor ?

dwaned, 2024-11-04:
Hi, thank you for replying. 
I have a Sonoff ZBDongle E. The sensor fw version is v1.00.53 (I did not trigger the OTA).

GuhSK5Fgt, 2024-11-06:
Same problem here.
SensorFW-Version 1.00.53, Sonoff ZBDongle E V7.4.4, Zigbee2MQTT1.41.0-1

RobLeighton67, 2024-11-13:
I've had issues with the device going offline for several hours at a time - I'm updating to 53 via OTA.

dwaned, 2024-12-02:
Updated to latest Zigbee2MQTT version 1.42.0-1, sensor version is v1.00.53, deleted and rejoined sensor, still no accelerometer values.

GuhSK5Fgt, 2025-01-04:
After updating to zigbee2mqtt version 2.0.0, all values are displayed again.

closed 2025-01-20 by dwaned