Koenkk/zigbee2mqtt #24586: pir SNZB-03 update 2zm version

rednet1, 2024-11-01:
### What happened?

after the last update the pir became a water leak sensor, in addition to which I found devices that had never been scanned

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

0000

### Adapter

sonoff E

### Setup

HA

sqorpo, 2024-11-02:
same issue after update. My automations don't know what to do with "leaking" vs "motion"

gebesys, 2024-11-02:
Same issue. PIR SNZB-03 is water leak sensor now. Related automations don’t work anymore. Downgraded so far…

Koenkk, 2024-11-03:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-03 by Koenkk