Koenkk/zigbee2mqtt #24588: [Wrong device]: SNZB-03

jbvs01, 2024-11-01:
### Link

https://de.aliexpress.com/item/1005002491782084.html?spm=a2g0n.productlist.0.0.26d4194c89Ndkw&browser_id=929a132181ba49fd9428ca6fa815c847&aff_trace_key=6090b0275b0f42d9ac620a6b336ef6cc-1729983692719-06740&aff_platform=msite&m_page_id=kfhcuzoxcavqxyug192e9142fa3e9386b46190c4d9&gclid=&pdp_npi=4%40dis%21EUR%2111.13%217.79%21%21%2111.83%218.28%21%40211b441e17304872672061026e41ac%2112000036233584005%21sea%21DE%21135908239%21X&algo_pvid=aabe6a54-b806-4156-a40b-cd4b8e625f85

### Model

RHK09

### Description

Motion Sensor

### Vendor

eWeLink

### Picture (link)

https://ae-pic-a1.aliexpress-media.com/kf/Sd21403d5a0674ce78fc3756aa6122ef8X/Ewelink-tuya-zigbee-bewegungs-sensor-smart-things-pr-senz-sensor-pir-sensor-arbeit-mit-zigbee-hub.jpg_640x640Q90.jpg_.webp

### Database entry

{"id":9,"type":"EndDevice","ieeeAddr":"0xa4c138ce9d5964f2","nwkAddr":33059,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x5cc7c1fffe45da20","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":28,"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x5cc7c1fffe45da20","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x5cc7c1fffe45da20","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-268385843},"lastSeen":1730483194328,"checkinInterval":3600}

### Notes

Sensor picture shows as water leak sensor after Z2M update to 1.41.0-1.
Also in details a water leak state is mentioned (which is not in the payload). - see attachment
Before the update it had the picture of the original SNZB-03 (instead of RHK09, which was ok for me)

[image]
[image]

cinemarene, 2024-11-01:
Same for me since upgrading to 1.41.0 today.

mrescher, 2024-11-01:
Hi, i have the same issue since the latest update.

sqorpo, 2024-11-02:
same issue after update

peteshoard, 2024-11-02:
Same issue

sphings79, 2024-11-02:
Same over here on two machines

mauriziosavio, 2024-11-02:
lo stesso, ho piu di 15 di questi dispositivi, speriamo risolvano

alray31, 2024-11-02:
@Koenkk 

I think this issue was caused by this change

https://github.com/Koenkk/zigbee-herdsman-converters/commit/5ff3e9f569011610c80eda9943b61c6d2e717fb5

from discussion in #24141

Leak detectors are now correctly showing as leak detectors but it made PIRs now also showing as leak detectors instead of PIRs

I dont have these devices but it looks like they are sharing similar figerprints

kitsunex3, 2024-11-02:
same issue  here ;)

mauriziosavio, 2024-11-02:
Ma è una cosa che si sistema con prossimo aggiornamento ho bisogna cambiare qualcosa nel codice?

Koenkk, 2024-11-02:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-02 by Koenkk