Koenkk/zigbee2mqtt #24590: Firmware is newer than latest firmware online

mathgim, 2024-11-01:
### What happened?

I got a new warning log in z2m about an old device I never had before.
Device: Heiman HS2WD-E
https://www.zigbee2mqtt.io/devices/HS2WD-E.html#heiman-hs2wd-e
```warning 2024-11-01 19:10:47zhc:ota:common: Firmware on '0x00047400002673az' ( Connected outlet) is newer than latest firmware online.```

### Zigbee2MQTT version

1.41.0 commit: unknown

### Adapter firmware version

20220219

### Adapter

usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Setup

Add-on on HA OS

Eoras, 2024-11-03:
I have same probleme with all my Legrand - 412170
Date du firmware : 20230911
Version du firmware : 0065

I did update last month ago, and now i have : 

```
warning 2024-11-03 20:15:46zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:16:11zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:16:19zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:16:30zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:16:46zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:16:54zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:16:56zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:01zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:01zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:07zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:09zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:11zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:11zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:11zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:13zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:16zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:20zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:32zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
warning 2024-11-03 20:17:35zhc:ota:common: Firmware on '' ( Teleruptor) is newer than latest firmware online.
```

titou4307, 2024-11-06:
Same problem...for me for 3 legrand

jlai79, 2024-11-12:
Same problem here with Tuya TS011F_plug_1

sbourdette, 2025-01-13:
Same for me

Nicosdem31, 2025-01-17:
Same issue on my side

glenharris, 2025-02-26:
Same issue here - could this be similar to #26439?  (I have the same experience of LeGrand devices being upgraded by Z2M, and now getting the 'newer firmware' warning, whereas the OTA firmware repository seems to show that the correct firmware is being checked)

closed 2025-05-05 by github-actions[bot]