Koenkk/zigbee2mqtt #24593: Tuya ZY-M100-24G ('_TZE204_ijxvkhd0') latest update problem

Dillton, 2024-11-01:
### What happened?

After latest update the values of sensitivity settings are presented as decimals in Z2M.
[image]

This causing a spam in HA log that the values should be different.
[image]

However in MQTT entities the values are correct.

### What did you expect to happen?

No log spam

### How to reproduce it (minimal and precise)

Update to latest release

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on in HAOS in Proxmox on Intel NUC

ippo112, 2024-11-02:
Hi,
I detect same error, 

_______________________________________________________
Registratore: homeassistant.components.mqtt.number
Fonte: components/mqtt/number.py:182
Integrazione: MQTT (documentazione, problemi)
Prima occorrenza: 00:00:32 (531 occorrenze)
Ultima registrazione: 06:53:41

Invalid value for number.presence_24ghz_occupancy_sensitivity: 20 (range 1.0 - 10.0)
Invalid value for number.presence_24ghz_occupancy_sensitivity: 80 (range 1.0 - 10.0)
Invalid value for number.presence_24ghz_motion_sensitivity: 30 (range 1.0 - 10.0)
Invalid value for number.presence_24ghz_motion_sensitivity: 100 (range 1.0 - 10.0)
Invalid value for number.presence_24ghz_occupancy_sensitivity: 100 (range 1.0 - 10.0)
________________________________________________

Adapter
SONOFF Zigbee 3.0 USB Dongle Plus-P
Versione Zigbee2MQTT
 [1.41.0]

Dillton, 2024-11-02:
More details in Koenkk/zigbee-herdsman-converters#8229

Koenkk, 2024-11-02:
I believe @ippo112 is working on a fix

magicus, 2024-11-03:
@Koenkk 

> I believe @ippo112 is working on a fix

That'd be awesome. There is some some discussion in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8229 about what has *not* worked. I get the impression that this Tuya device is doing something very odd, so it might need a fix on a higher level to deal with this behavior.

magicus, 2024-11-04:
Partial revert posted in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8272.

closed 2025-01-11 by github-actions[bot]