Koenkk/zigbee2mqtt #24595: error : z2m: No converter available for 'action'

jglafkid, 2024-11-01:
### What happened?

I get that error on z2m since months : error : z2m: No converter available for 'action'
no way to identify where it comes from ! which device causes this action. 

### What did you expect to happen?

no error

### How to reproduce it (minimal and precise)

I don't know.
conf :
Version de Zigbee2MQTT : [1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2) commit: [e06848db](https://github.com/Koenkk/zigbee2mqtt/commit/e06848db)
Type de coordinateur : ConBee2/RaspBee2
Révision du coordinateur : 0x26780700
Adresse IEEE du Coordinateur :0x00212effff0
Version de l'interface : 0.7.4
Version Zigbee-herdsman-converters : 20.21.0
Version Zigbee-herdsman : 2.1.3
Statistiques
Total 83
Par type d'appareil
Appareils terminaux: 45
Routeurs: 38
Par source d'alimentation
Secteur (monophasé): 57
Batterie: 22
undefined: 4
Par vendeur
LUMI: 41
_TZ3000_gjnozsaz: 23
_TZ3000_qeuvnohg: 9
HEIMAN: 2
_TZ3000_2putqrmw: 2
Philips: 1
LiXee: 1
MultIR: 1
OWON: 1
frient A/S: 1
undefined: 1
Par modèle
TS011F: 34
lumi.switch.b1lc04: 20
lumi.remote.acn003: 5
lumi.weather: 4
lumi.switch.b2nc01: 3
lumi.sensor_switch.aq3: 3
lumi.switch.b2lc04: 2
lumi.remote.acn004: 2
lumi.sensor_magnet.aq2: 2
SML001: 1
lumi.switch.b1nc01: 1
ZLinky_TIC: 1
52debf035a1b4a66af56415474646c02: 1
THS317-ET: 1
COSensor-EF-3.0: 1
SmokeSensor-EM: 1
HESZB-120: 1

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

Debian GNU/Linux 11 (bullseye) 64bits (aarch64), jeedom on Rpi4

### Debug log

error 2024-11-01 22:19:08z2m: No converter available for 'action' ("")

clockbrain, 2024-11-03:
My suspicion is that z2m is reporting that error when it receives an empty mqtt message from Home Assistant. I can get the exact same error message by sending the following mqtt message to z2m:
```
zigbee2mqtt/Back garden light/set
{
  "action": ""
}
```
I suggest you monitor your mqtt traffic using e.g. MQTT Explorer for Windows or similar.

jglafkid, 2024-11-03:
I am using jeedomi will try to monitor mqtt traffic. it must be si stalled on my pi as both jeedom and mqtt are on the same rpi ?normally jeedom retrieve available commands from mqtt.

clockbrain, 2024-11-03:
No, it doesn't have to be on the same pi as z2m. I have z2m on my pi, home assistant and mosquitto on my Synology, and I monitor traffic from my Windows desktop.

As long as you configure the mqtt monitor with the IP address, username and password of your mqtt broker you will be able to observe message traffic.

Options include:
https://pypi.org/project/mqtt-cli/
https://mqtt-explorer.com/
https://mqttx.app/

jglafkid, 2024-11-03:
I installed explorer then get the logs but is is still not understandable for me :

{"level":"error","message":"z2m: No converter available for 'action' (\"\") »}

zigbee2mqtt/bridge

zigbee2mqtt/bridge/info : 

{"commit":"e06848db","config":{"advanced":{"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":false,"channel":20,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"last_seen":"disable","legacy_api":true,"legacy_availability_payload":true,"log_debug_namespace_ignore":"","log_debug_to_mqtt_frontend":false,"log_directory":"/var/www/html/plugins/zigbee2mqtt/data/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.log","log_level":"error","log_namespaced_levels":{},"log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":53902,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"availability":{},"blocklist":[],"device_options":{},"devices":{"0x000d6f00162bea86":{"description":"0x000d6f00162bea86 détecteur fumée ","friendly_name":"Détecteur fumée Helman"},"0x00158d0005d29ee2":{"description":"0x00158d0005d29ee2 lixee linky","friendly_name":"lixee linky"},"0x00158d0006d6bc69":{"description":"0x00158d0006d6bc69 thermomètre interieur","friendly_name":"Thermomètre chambre 20"},"0x00158d0006d6bd95":{"description":"0x00158d0006d6bd95","friendly_name":"capteur temp petit salon"},"0x00158d00074adf91":{"description":"0x00158d00074adf91 Bouton 22","friendly_name":"Bouton B-22 Petit Salon"},"0x00158d0007bcccc9":{"description":"0x00158d0007bcccc9  Bouton 20","friendly_name":"Bouton B-20"},"0x00158d0007bcd8e5":{"description":"0x00158d0007bcd8e5 Bouton 21","friendly_name":"Bouton B-21 Grand Salon"},"0x00158d000890d66b":{"description":"0x00158d000890d66b capteur cuisine","friendly_name":"Fenêtre Cuisine"},"0x00158d0008912983":{"description":"0x00158d0008912983","friendly_name":"porte entrée"},"0x00158d0009cbf30a":{"description":"0x00158d0009cbf30a capteur temp","friendly_name":"température 2e"},"0x00158d0009cbfc82":{"description":"0x00158d0009cbfc82 capteur de température","friendly_name":"Temp couloir 1er"},"0x0015bc00340015fe":{"description":"0x0015bc00340015fe détecteur de feu studio","friendly_name":"détecteur feu studio"},"0x00178801032b3c6d":{"description":"0x00178801032b3c6d","friendly_name":"Detecteur de mouvement"},"0x385b44fffe95c4fd":{"description":"0x385b44fffe95c4fd","friendly_name":"Ballon SSol Thermodynamique"},"0x385b44fffe95c5fa":{"description":"0x385b44fffe95c5fa","friendly_name":"STD Disj prise chauffage"},"0x385b44fffe95c705":{"friendly_name":"SDE 2 Seche serviette"},"0x385b44fffe95c7aa":{"description":"0x385b44fffe95c7aa","friendly_name":"STD Ballon"},"0x385b44fffe95c879":{"description":"0x385b44fffe95c879","friendly_name":"Lumieres exterieures chateau"},"0x385b44fffe95c936":{"friendly_name":"Lampe Palier automatique"},"0x385b44fffe95d17c":{"description":"0x385b44fffe95d17c","friendly_name":"STD SDE seche serviette"},"0x385b44fffe95d375":{"description":"0x385b44fffe95d375","friendly_name":"STD Disj lumiere cuisine"},"0x385b44fffe95d3ee":{"friendly_name":"Prise chauffage 22"},"0x3c6a2cfffed3afb1":{"description":"0x3c6a2cfffed3afb1 thermo piscine","friendly_name":"thermomètre piscine"},"0x54ef4410003ffdbf":{"description":"0x54ef4410003ffdbf","friendly_name":"STD SDE Lumieres"},"0x54ef4410004f7b6e":{"description":"0x54ef4410004f7b6e inter sans fil 2e escalier /coiloir","friendly_name":"Inter SF 2eme"},"0x54ef4410004f885f":{"description":"0x54ef4410004f885f lumiere 22","friendly_name":"Lumiere chambre 22"},"0x54ef441000512508":{"description":"0x54ef441000512508 Interrupteur sans fil petit et grand salon","friendly_name":"Inter SF salons"},"0x54ef441000565292":{"description":"0x54ef441000565292","friendly_name":"interrupteurs entrée nord"},"0x54ef4410005977e6":{"description":"0x54ef4410005977e6","friendly_name":"Lustres SaM"},"0x54ef44100059a107":{"description":"0x54ef44100059a107","friendly_name":"Plafonnier Cuisine"},"0x54ef4410005b0310":{"description":"0x54ef4410005b0310 inter SDE 2e","friendly_name":"inter SDE 2e"},"0x54ef4410005c349f":{"description":"0x54ef4410005c349f","friendly_name":"lumière cave"},"0x54ef4410005c34f4":{"description":"0x54ef4410005c34f4","friendly_name":"STD Plafonnier Ilot"},"0x54ef4410005c35cc":{"description":"0x54ef4410005c35cc","friendly_name":"STD Lumiere exterieur Ouest"},"0x54ef4410005c3703":{"description":"0x54ef4410005c3703","friendly_name":"Lumiere Couloir 1er"},"0x54ef4410005c38fa":{"friendly_name":"Ou est il 0x54ef4410005c38fa"},"0x54ef4410005c38fd":{"description":"0x54ef4410005c38fd","friendly_name":"Lustre Entree"},"0x54ef4410005c39a4":{"description":"0x54ef4410005c39a4 lampe lavabo ch 12","friendly_name":"Lumière lavabo chambre 12"},"0x54ef4410005c3a46":{"description":"0x54ef4410005c3a46","friendly_name":"STD Lumiere exterieur Nord"},"0x54ef4410005c3f32":{"description":"0x54ef4410005c3f32 Couloir 2e","friendly_name":"lumiere Couloir 2e"},"0x54ef4410005c4013":{"description":"0x54ef4410005c4013","friendly_name":"STD SDE Ventilateur"},"0x54ef4410005c4015":{"description":"0x54ef4410005c4015","friendly_name":"Lumiere escalier"},"0x54ef4410005c401e":{"description":"0x54ef4410005c401e plafonnier grenier","friendly_name":"Plafonnier Grenier"},"0x54ef4410005c4310":{"description":"0x54ef4410005c4310 prise pilotée ch11","friendly_name [... cut ...] tus","type":32},{"name":"groupid","type":33}]},"getMembershipRsp":{"ID":2,"parameters":[{"name":"capacity","type":32},{"name":"groupcount","type":32},{"name":"grouplist","type":1002}]},"removeRsp":{"ID":3,"parameters":[{"name":"status","type":32},{"name":"groupid","type":33}]},"viewRsp":{"ID":1,"parameters":[{"name":"status","type":32},{"name":"groupid","type":33},{"name":"groupname","type":66}]}}},"genIdentify":{"ID":3,"attributes":{"identifyCommissionState":{"ID":1,"type":255},"identifyTime

￼

jglafkid, 2025-01-03:
Any solution ?

clockbrain, 2025-01-03:
I can't tell anything from those mqtt explorer logs. They are only partial - each is truncated. Also, they seem to be just the logs from zigbee2mqtt. I expect the problem to show in the message topics from Jeedom to zigbee2mqtt.
As per https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html the device messages will be under topics such as _zigbee2mqtt/FRIENDLY_NAME/set_
But the problem is we don't know which device is sending that empty action message.

jglafkid, 2025-01-03:
Thanx. I agree with the pb that we don't know which device is sending that empty action message
What log do you need ? 
I have around 90 devices so I can’t check manually all YAML to find the one (if any) that have a name and an action Null. 

Is there a folder where I can see all existing YAML with the terminal or by SSH ? 

Regards

clockbrain, 2025-01-03:
Ok, I've looked deeper and I may be wrong in assuming its an incoming mqtt message. Zigbee2mqtt can show that error message for either an incoming mqtt message or for a device generated event.
Since the error message doesn't say which device you will need to turn on 'debug' level logging in zigbee2mqtt and identify the device through earlier log messages around the same timestamp as the no converter error.

You can set log level to debug through the frontend or via the log_level option in yaml https://www.zigbee2mqtt.io/guide/configuration/logging.html

closed 2025-03-12 by github-actions[bot]

tammeryousef1006, 2025-03-21:
not stale

tammeryousef1006, 2025-03-21:
[2025-03-19 01:24:29] error: 	z2m: No converter available for 'state' ("")
[2025-03-19 01:37:16] error: 	z2m: No converter available for 'state' ("")
[2025-03-19 01:38:07] error: 	z2m: No converter available for 'state' ("")
[2025-03-20 01:12:03] error: 	z2m: No converter available for 'state' ("")
[2025-03-20 03:18:28] error: 	z2m: No converter available for 'state' ("")

clockbrain, 2025-04-04:
@jglafkid, @tammeryousef1006 can you test again with zigbee2mqtt version 2.2.0.

This version includes improved logging to identify the device for which it can't find a converter -  if my assumption about which error is occurring is correct.

tammeryousef1006, 2025-04-04:
@clockbrain i confirm with latest update i don't have this issue anymore , thanks a lot

clockbrain, 2025-04-05:
@tammeryousef1006 just to be clear, my patch didn't fix anything. I just improved the logging to help identify which device the error message related to. Its possible though that other fixes in this release did indeed fix the problem.

Let us know if that error message shows up again.

tammeryousef1006, 2025-04-05:
sure , will keep you updated

ignidis, 2026-10-24:
Well, i have the same issue since an update some months ago ... I did open a case which got no attention and after some months of research i found this is related to the state of devices which got an OTA update earlier ... All of these devices have an attribute "update" in their state an every time they send and state update the error is posted to the log.
New device which are either not OTA updatable or have not entry in the OTA tag do not have the "upadte" attribute in the state and do not cause any trouble.

Example of state for a device throwing errors ...
{
    "brightness": 243,
    "color_mode": "color_temp",
    "color_temp": 250,
    "last_seen": "2026-10-24T15:42:19.727Z",
    "level_config": {
        "on_level": null
    },
    "linkquality": 240,
    "power_on_behavior": "on",
    "state": "ON",
    "update": {
        "installed_version": 587757105,
        "latest_release_notes": "https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html",
        "latest_source": "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/IKEA/tradfri-driver-hp_release_prod_v587757105_b42c57bf-cb9c-478c-8327-5812a3286a64.ota",
        "latest_version": 587757105,
        "state": "idle"
    },
    "color_options": null,
    "effect": null,
    "identify": null
}
example of a "clean" device
{
    "brightness": 0,
    "color_mode": "color_temp",
    "color_temp": 153,
    "last_seen": "2026-10-24T15:52:47.560Z",
    "linkquality": 176,
    "power_on_behavior": "on",
    "state": "OFF",
    "effect": null,
    "identify": null
}

My current approach (while i try to learn how to create an external converter) is to filter the "update" attribute from the state ... This will take some time as i have quite a few settings for devices to modify ... I'll post again when i get all the states cleaned and see if this does help