Koenkk/zigbee2mqtt #24596: Interview fails when trying to pair a Bosch BSD-2 with Sonoff Dongle-E

cyr0n, 2024-11-01:
### What happened?

I am trying to pair a Bosch BSD-2 smoke detector with my Sonoff Dongle-E coordinator.

While initial pairing succeeds, afterwards the interview fails with the following messages given in the log:

info 2024-11-01 22:28:19z2m: Device '0xf4b3b1fffe6d9f57' joined
info 2024-11-01 22:28:19z2m: Starting interview of '0xf4b3b1fffe6d9f57'
info 2024-11-01 22:28:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4b3b1fffe6d9f57","ieee_address":"0xf4b3b1fffe6d9f57"},"type":"device_joined"}'
info 2024-11-01 22:28:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4b3b1fffe6d9f57","ieee_address":"0xf4b3b1fffe6d9f57","status":"started"},"type":"device_interview"}'
error 2024-11-01 22:28:54zh:controller: Interview failed for '0xf4b3b1fffe6d9f57 with error 'Error: Interview failed because can not get node descriptor ('0xf4b3b1fffe6d9f57')'
error 2024-11-01 22:28:54z2m: Failed to interview '0xf4b3b1fffe6d9f57', device has not successfully been paired

I tried re-pairing it several times, with restarts of Home Assistants and completely deleting the device from Zigbee2MQTT inbetween, with no success. Other sensors (mainly Aqara, I don't have any other Bosch devices) are working fine.

I am using the following versions:

Zigbee2MQTT version: 1.41.0 commit: unknown
Coordinator type: EmberZNet
Coordinator revision: 7.4.4 [GA]
zigbee-herdsman version: 2.1.7

I can find a lot of comments here in the repository that people have been able to successfully pair their BSD-2s, so if anyone could give me a hint on what might be going wrong or what I could try, I would be very grateful.

Thanks!

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.4

### Adapter

Sonoff Dongle-E

### Setup

RaspberryPi Model 4 with Home Assistant OS, as an add-on

### Debug log

[log.log](https://github.com/user-attachments/files/17604423/log.log)

closed 2025-01-09 by github-actions[bot]

Sailing74, 2025-01-17:
Did you ever solve this? Having the same issue hier with z2m 2.0.0 and 2 out of 8 BSD-2s...

cyr0n, 2025-02-09:
Sorry, I'm not sure I can help you much.

For me, the issue resolved itself after a while. I performed a factory reset of the smoke alarm, then removed the battery and let it sit for a while just to be sure - although I'm not sure if that was necessary.

After a week or two I added the install code to my Home Assistant again and suddenly it paired without a hitch. I'm sure though that I used the correct install code during the first tries too, so I'm not sure what made the difference the second time.