Koenkk/zigbee2mqtt #24597: Moes MS-108ZR - Inverted percentage in position

Llenas09, 2024-11-02:
### What happened?

After updating to version 1.41.0 commit: cc31ceab, the Moes MS-108ZR device has started to display the percentage in "Position" in reverse.
Now when the blind is open it shows a 0 and when it is closed it shows a 100. The button UP and Down works correctly.

Before the update it was the other way around. 0 when it is closed, 100 when it is open. This is how they are displayed in other models of blind controllers that I have.

### What did you expect to happen?

when the blind is open shows a 100 and when it is closed it shows a 0.

### How to reproduce it (minimal and precise)

Open and close the blind.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Setup

Proxmox VE Zigbee2MQTT LXC

LatchyRed, 2024-11-02:
Same problem with Meos MS-108ZR and conbee II

rogervdh, 2024-11-02:
Same issue here since update to 1.41.0.... MS-108ZR wth SLB-06.

MrAges, 2024-11-02:
Same Problem since update today: Sonoff ZBDongle-E and MS-108ZR

Lloydchabert, 2024-11-02:
Same here with Conbee II.

Lloydchabert, 2024-11-02:
Found the solution.
In the configuration.yaml of zigbee2mqtt folder, there is invert_cover value along your MS-108ZR devices that needs to be set to "true". Restart z2m and you should be good. if missing just paste     `invert_cover: true`

LatchyRed, 2024-11-02:
Yes it works !
[image]

Koenkk, 2024-11-03:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-03 by Koenkk

luisrosety, 2024-12-04:
The same problem occurred, and this is worrying.
A few weeks ago, the same thing happened.
The invert_cover value was set to false, and switching it to true corrected the problem.
Today, it happened again, and the solution was switching it back to false.
This is quite concerning because the automation opened the curtains when they should have been closed.
Relying on this automation when being away becomes a safety risk.