Koenkk/zigbee2mqtt #24601: [New device support]: Zigbee阀控超声波水表

MiAutomations, 2024-11-02:
### Link

Not available yet

### Database entry

{"id":187,"type":"EndDevice","ieeeAddr":"0xa4c138ebc5191988","nwkAddr":27195,"manufId":4417,"manufName":"_TZE200_vuwtqx0t","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"appVersion":72,"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1730550136997}

### Zigbee2MQTT version

1.41.0

### Comments

I tried to add this device without success for now, it appears as not supported.

Please let me know if you need any further information more 

Thank you 

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_vuwtqx0t',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

All the functions

 Empty exposes definition

mrmaximas, 2024-11-03:
what about post of link to the device? what about picture?

MiAutomations, 2024-11-03:
> what about post of link to the device? what about picture?

Photo if need I can take and share, regarding the link this device was delivery by hand, but I will try to get a link...

mrmaximas, 2024-11-04:
> > what about post of link to the device? what about picture?
> 
> Photo if need I can take and share, regarding the link this device was delivery by hand, but I will try to get a link...

https://github.com/Koenkk/zigbee2mqtt/issues/21255

MiAutomations, 2025-01-04:
The issue continues 

Please keep this topic opened 

Thank you

closed 2025-03-14 by github-actions[bot]