Koenkk/zigbee2mqtt #24604: Unable to install via OTA the firmware update available for 3reality smart-soil-moisture-sensor

MiAutomations, 2024-11-02:
### What happened?

Hello I received the "warning" to install/availability of a new firmware fo the device  3reality smart-soil-moisture-sensor, but when I try I always receiving the same msg 

[image]

[image]

I try to change the battery but i still facing the same issue  the firmware still not installed, can you please help me in odrer to try solve this issue? 

### What did you expect to happen?

I expect a success firmware update 

### How to reproduce it (minimal and precise)

Try to install the firmware 

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

slzb 06

### Setup

Z2M addon

mrmaximas, 2024-11-02:
Please activate device 1s before update.

MiAutomations, 2024-11-02:
> ore update.

How can I do this?

Great-Chart, 2024-11-03:
Not sure for this device but for end devices a push of the reset button (not as much as for pairing) or forcing it to update (wave at pir etc) will bring it out of sleep mode.
Anything to wake this up so it communicates to HA

mrmaximas, 2024-11-03:
> > ore update.
> 
> How can I do this?

short press to pair button

MiAutomations, 2024-11-03:
> Please activate device 1s before update.

Thank you this work for me, I press the button when I execute the OTA update

closed 2025-01-10 by github-actions[bot]