Koenkk/zigbee2mqtt #24605: [New device support]: Moes Curtain Robot ADCBZI01

and7ey, 2024-11-02:
### Link

https://aliexpress.ru/item/1005007923066724.html

### Database entry

not found there :(

### Zigbee2MQTT version

1.22.1

### Comments

The device 0xa4c138df4695086d is detected as https://www.zigbee2mqtt.io/devices/TS030F.html, but doesn't work as expected. Sometimes it moves the curtain by 10 cms only. But in many cases just doesn't react and returns timeout.

 Here are some logs:
 ```
 info 2024-11-02 17:34:20zh:controller: Interview for '0xa4c138df4695086d' started
info 2024-11-02 17:34:20z2m: Device '0xa4c138df4695086d' joined
info 2024-11-02 17:34:20z2m: Starting interview of '0xa4c138df4695086d'
info 2024-11-02 17:34:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138df4695086d","ieee_address":"0xa4c138df4695086d"},"type":"device_joined"}'
info 2024-11-02 17:34:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138df4695086d","ieee_address":"0xa4c138df4695086d","status":"started"},"type":"device_interview"}'
info 2024-11-02 17:34:41zh:controller:device: Device '0xa4c138df4695086d' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
info 2024-11-02 17:34:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138df4695086d","ieee_address":"0xa4c138df4695086d"},"type":"device_announce"}'
info 2024-11-02 17:34:54zh:controller: Succesfully interviewed '0xa4c138df4695086d'
info 2024-11-02 17:34:54z2m: Successfully interviewed '0xa4c138df4695086d', device has successfully been paired
info 2024-11-02 17:34:54z2m: Device '0xa4c138df4695086d' is supported, identified as: Tuya Smart blind controller (TS030F)
info 2024-11-02 17:34:54z2m: Configuring '0xa4c138df4695086d'
info 2024-11-02 17:34:54z2m:mqtt: MQTT publish: topic 'homeassistant/cover/0xa4c138df4695086d/cover/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c138df4695086d/set","device":{"identifiers":["zigbee2mqtt_0xa4c138df4695086d"],"manufacturer":"Tuya","model":"Smart blind controller (TS030F)","name":"0xa4c138df4695086d","via_device":"zigbee2mqtt_bridge_0x00124b002a2e94e2"},"name":null,"object_id":"0xa4c138df4695086d","origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0xa4c138df4695086d","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0xa4c138df4695086d/set","state_closed":"CLOSE","state_open":"OPEN","state_stopped":"STOP","state_topic":"zigbee2mqtt/0xa4c138df4695086d","unique_id":"0xa4c138df4695086d_cover_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info 2024-11-02 17:34:54z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart blind controller","exposes":[{"features":[{"access":3,"label":"State","name":"state","property":"state","type":"enum","values":["OPEN","CLOSE","STOP"]},{"access":7,"description":"Position of this cover","label":"Position","name":"position","property":"position","type":"numeric","unit":"%","value_max":100,"value_min":0}],"type":"cover"},{"access":2,"label":"Border","name":"border","property":"border","type":"enum","values":["up","down","up_delete","down_delete"]},{"access":7,"label":"Calibration time","name":"calibration_time","property":"calibration_time","type":"numeric","value_max":100,"value_min":0},{"access":7,"description":"Reverse the motor, resets all endpoints! Also the upper border after hardware initialisation. Be careful!After this you have to turn off and turn on the roller so that it can drive into the uppest position.","label":"Motor reversal","name":"motor_reversal","property":"motor_reversal","type":"binary","value_off":"OFF","value_on":"ON"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS030F","options":[{"access":2,"description":"Inverts the cover position, false: open=100,close=0, true: open=0,close=100 (default false).","label":"Invert cover","name":"invert_cover","property":"invert_cover","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Do not publish set cover target position as a normal 'position' value (default false).","label":"Cover position tilt disable report","name":"cover_position_tilt_disable_report","property":"cover_position_tilt_disable_report","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c138df4695086d","ieee_address":"0xa4c138df4695086d","status":"successful","supported":true},"type":"device_interview"}'
info 2024-11-02 17:34:54z2m:mqtt: MQTT publish: topic 'homeassistant/select/0xa4c138df4695086d/border/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c138df4695086d/set/border","device":{"identifiers":["zigbee2mqtt_0xa4c138df4695086d"],"manufacturer":"Tuya","model":"Smart blind controller (TS030F)","name":"0xa4c138df4695086d","via_device":"zigbee2mqtt_bridge_0x00124b002a2e94e2"},"enabled_by_default":true,"name":"Border","object_id":"0xa4c138df4695086d_border","options":["up","down","up_delete","down_delete"],"origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"unique_id":"0xa4c138df4695086d_border_zigbee2mqtt"}'
info 2024-11-02 17:34:54z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138df4695086d/calibration_time/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138df4695086d"],"manufacturer":"Tuya","model":"Smart blind controller (TS030F)","name":"0xa4c138df4695086d","via_device":"zigbee2mqtt_bridge_0x00124b002a2e94e2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:wrench-clock","name":"Calibration time","object_id":"0xa4c138df4695086d_calibration_time","origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c138df4695086d","unique_id":"0xa4c138df4695086d_calibration_time_zigbee2mqtt","value_template":"{{ value_json.calibration_time }}"}'
error 2024-11-02 17:34:54z2m: Failed to configure '0xa4c138df4695086d', attempt 1 (Error: Bind 0xa4c138df4695086d/1 genPowerCfg from '0x00124b002a2e94e2/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:9138:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
info 2024-11-02 17:34:54z2m:mqtt: MQTT publish: topic 'homeassistant/number/0xa4c138df4695086d/calibration_time/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c138df4695086d/set/calibration_time","device":{"identifiers":["zigbee2mqtt_0xa4c138df4695086d"],"manufacturer":"Tuya","model":"Smart blind controller (TS030F)","name":"0xa4c138df4695086d","via_device":"zigbee2mqtt_bridge_0x00124b002a2e94e2"},"entity_category":"config","icon":"mdi:wrench-clock","max":100,"min":0,"name":"Calibration time","object_id":"0xa4c138df4695086d_calibration_time","origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c138df4695086d","unique_id":"0xa4c138df4695086d_calibration_time_zigbee2mqtt","value_template":"{{ value_json.calibration_time }}"}'
info 2024-11-02 17:34:54z2m:mqtt: MQTT publish: topic 'homeassistant/switch/0xa4c138df4695086d/switch_motor_reversal/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c138df4695086d/set/motor_reversal","device":{"identifiers":["zigbee2mqtt_0xa4c138df4695086d"],"manufacturer":"Tuya","model":"Smart blind controller (TS030F)","name":"0xa4c138df4695086d","via_device":"zigbee2mqtt_bridge_0x00124b002a2e94e2"},"entity_category":"config","icon":"mdi:arrow-l [... cut ...]
const {windowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS030F'],
    model: 'TS030F',
    vendor: '_TZ3210_sxtfesc6',
    description: 'Automatically generated definition',
    extend: [windowCovering({"controls":["lift","tilt"]})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

doesn't work as expected. Sometimes it moves the curtain by 10 cms only. But in many cases just doesn't react and returns timeout.

and7ey, 2024-11-02:
Here some of the data points found:

```
[2024-11-02 20:42:56] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":110}],"seq":29952}' from endpoint 1 with groupID 0
[2024-11-02 20:42:56] debug:    zhc:tuya: Datapoint 110 not defined for '_TZ3210_sxtfesc6' with value 0

[2024-11-02 20:43:09] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,16],"type":"Buffer"},"datatype":2,"dp":107}],"seq":30464}' from endpoint 1 with groupID 0
[2024-11-02 20:43:09] debug:    zhc:tuya: Datapoint 107 not defined for '_TZ3210_sxtfesc6' with value 16

[2024-11-02 20:43:09] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,16],"type":"Buffer"},"datatype":2,"dp":107}],"seq":30720}' from endpoint 1 with groupID 0
[2024-11-02 20:43:09] debug:    zhc:tuya: Datapoint 107 not defined for '_TZ3210_sxtfesc6' with value 16

[2024-11-02 20:51:56] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,52],"type":"Buffer"},"datatype":2,"dp":107}],"seq":33536}' from endpoint 1 with groupID 0
[2024-11-02 20:51:56] debug:    zhc:tuya: Datapoint 107 not defined for '_TZ3210_sxtfesc6' with value 52

[2024-11-02 20:43:09] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":13}],"seq":30976}' from endpoint 1 with groupID 0
[2024-11-02 20:43:09] debug:    zhc:tuya: Datapoint 13 not defined for '_TZ3210_sxtfesc6' with value 100

[2024-11-02 20:51:56] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":3}],"seq":33280}' from endpoint 1 with groupID 0
[2024-11-02 20:51:56] debug:    zhc:tuya: Datapoint 3 not defined for '_TZ3210_sxtfesc6' with value 100

[2024-11-02 20:50:12] debug:    z2m: Received Zigbee message from '0xa4c138df4695086d', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":13}],"seq":32000}' from endpoint 1 with groupID 0
[2024-11-02 20:50:12] debug:    zhc:tuya: Datapoint 13 not defined for '_TZ3210_sxtfesc6' with value 0
```

jmijn, 2024-11-03:
Following this bc i want to buy these devices too, but only with z2m support 😉

jmijn, 2024-11-08:
@and7ey have you made any progress?

and7ey, 2024-11-08:
With [dinamik1488](https://github.com/dinamik1488) support we've developed more or less working version. But I've ordered tuya zigbee gateway to make the integration better.

and7ey, 2024-11-16:
Here is the current version

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: tuya.fingerprint('TS030F', ['_TZ3210_sxtfesc6']),
        model: 'ADCBZI01',
        vendor: 'Moes',
        description: 'Curtain Robot',
        fromZigbee: [fz.cover_position_tilt, tuya.fz.datapoints],
        toZigbee: [tz.cover_position_tilt, tz.cover_state, tuya.tz.datapoints],
        exposes: [
            e.cover_position(),
            e.position(),
            e.battery(),
            e.illuminance(),
        ],
        meta: {
            tuyaDatapoints: [
                [3, 'position', tuya.valueConverter.raw],
                [13, 'battery', tuya.valueConverter.raw],
                [107, 'illuminance', tuya.valueConverter.raw],                      
              ],
        },
};
    
module.exports = definition;
```

I can open/close the curtain and control the position. The battery and illuminance are available.
But I can not revert the motor ;(

I've got the following table from MOES:
[image]

Would appreciate if somebody help me with motor inversion.

matthewharrington, 2024-11-27:
Following as I have this device as well. Not sure how I can help.

and7ey, 2024-11-27:
@matthewharrington the converter above works more or less. 

You can experiment more with the device itself. Somehow you can control the motor inversion with the button on the device. If you double click it, it moves left (or right), with triple click it moves in the opposite direction. The single click remembers the position. We should investigate how to control the initial movement (left or right). I have several devices and one moves left and another moves right, but I am not sure how I did it

mrmaximas, 2024-11-28:
@and7ey after getting a tuya hub, make an account at iot.tuya.com and see which DPs will change with which actions.
It will become clearer.

and7ey, 2024-11-28:
> see which DPs will change with which actions

It didn't help, unfortunately. Something works, but not everything (for ex., motor inversion or opening/closure, when dps are used directly).

matthewharrington, 2024-11-30:
Thanks for your efforts here @and7ey These motors have me perplexed. I have managed to get them going with your config. I have two without a Tuya hub.

I am using cover position to open and close them. The cover position reporting seems to be quite random. Sometimes one of them says 100 and other times it says 0 even though it is in the same position.

papka1974, 2024-12-21:
Bought Moes ZC-RB01-A-EN. Unable to control neither with default config, nor using provided external configuration. Any help on this model?

vasilevrv, 2025-01-20:
> Here is the current version
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>         fingerprint: tuya.fingerprint('TS030F', ['_TZ3210_sxtfesc6']),
>         model: 'ADCBZI01',
>         vendor: 'Moes',
>         description: 'Curtain Robot',
>         fromZigbee: [fz.cover_position_tilt, tuya.fz.datapoints],
>         toZigbee: [tz.cover_position_tilt, tz.cover_state, tuya.tz.datapoints],
>         exposes: [
>             e.cover_position(),
>             e.position(),
>             e.battery(),
>             e.illuminance(),
>         ],
>         meta: {
>             tuyaDatapoints: [
>                 [3, 'position', tuya.valueConverter.raw],
>                 [13, 'battery', tuya.valueConverter.raw],
>                 [107, 'illuminance', tuya.valueConverter.raw],                      
>               ],
>         },
> };
>     
> module.exports = definition;
> ```
> 
> I can open/close the curtain and control the position. The battery and illuminance are available. But I can not revert the motor ;(
> 
> I've got the following table from MOES: <img alt="image" width="932" src="https://private-user-images.githubusercontent.com/900474/386850456-d01d2358-443b-46e2-bf98-36cc0ae5d636.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzczNDc1OTEsIm5iZiI6MTczNzM0NzI5MSwicGF0aCI6Ii85MDA0NzQvMzg2ODUwNDU2LWQwMWQyMzU4LTQ0M2ItNDZlMi1iZjk4LTM2Y2MwYWU1ZDYzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEyMFQwNDI4MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04Mjc2NTlhNzk0YjNjZDcyNDg3NTk5ZDVmZmYxZTQ5NjM0NzMxMWI4Y2RmYjRkODc2NDMwN2RjNGU5YzNkMjVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.qThiUygqYkT9FL5AHbui1jl_P4FDdKL__UyCsA88nNI">
> 
> Would appreciate if somebody help me with motor inversion.

Hello @and7ey ! It works for me, but there are errors, the position and brightness do not always match. Did you manage to complete the work to support this device?

Thanks in advance!

vasilevrv, 2025-01-20:
Hm.. and now it seems to be ok. This error was fixed by change the "Illuminance calibration" setting to 0.

sneakythr0ws, 2025-01-31:
guys! how to calibrate it without tuya gateway?(with z2m only)

and7ey, 2025-01-31:
> guys! how to calibrate it without tuya gateway?(with z2m only)

use the button on the device itself

sneakythr0ws, 2025-01-31:
> > guys! how to calibrate it without tuya gateway?(with z2m only)
> 
> use the button on the device itself

set in closed position and double press? right?

and7ey, 2025-01-31:
> set in closed position and double press? right?
I haven't found the exact logic - you will have to experiment. Double click moves it into one direction, triple click to another and single click stops it.

zolo2hin, 2025-02-15:
Guys, just the information: I`ve created some models for this motor to handle two curtains with the single motor: https://www.printables.com/model/1193078-moes-curtains-motor-adapter

sneakythr0ws, 2025-02-15:
> Guys, just the information: I`ve created some models for this motor to handle two curtains with the single motor: https://www.printables.com/model/1193078-moes-curtains-motor-adapter Can be easily 3d-printed and assembled

Great job!
Btw maybe you know how to calibrate it?)

zolo2hin, 2025-02-15:
> > > guys! how to calibrate it without tuya gateway?(with z2m only)
> > 
> > 
> > use the button on the device itself
> 
> set in closed position and double press? right?

I`ve done like that. 2 and 3 presses work fine for me, changes the direction.

WRLPDZ, 2025-03-25:
Hi @and7ey,

I appreciate your work

Question: where should I paste this code to make those motors fully operational in Z2M.

I can open, close, also manually calibrated thank to @zolo2hin instruction.

But illumination would be grate for automations and battery for maintenance.

Thanks

and7ey, 2025-03-27:
> where should I paste this code to make those motors fully operational in Z2M

https://www.zigbee2mqtt.io/advanced/more/external_converters.html

`External converters are stored in data/external_converters folder`

Create a new file there.

WRLPDZ, 2025-03-29:
@and7ey Thank you it's working.

Will you create a pull request to make it officially supported?

hansons83, 2025-04-18:
Hi. Have anybody got it working on Zigbee2MQTT version 2.2.1? I keep getting below errors
`[2025-04-18 11:50:32] error: 	z2m: Failed to load external extension 'robot.js'. Check the code for syntax error and make sure it is up to date with the current Zigbee2MQTT version.
[2025-04-18 11:50:32] error: 	z2m: Invalid external extension 'robot.js' was ignored and renamed to prevent interference with Zigbee2MQTT.`

and7ey, 2025-06-25:
up

fahdr, 2025-07-07:
Following this thread for progress. I have 8 of these. Don't want to use the tuya hub.

benlangfield, 2025-07-13:
I purchased one of these to test out, judging by this comment thread it may not be worth the trouble.  Might just try the more expensive Aqara one instead, looks like that's the only one officially supported by Zigbee2MQTT

Slejpnir, 2025-07-13:
I have a device with the same fingerprint but it's not working at all with the proposed converter. Any ideas why? Using z2m 2.5.1.
Thanks in advance for any help.

fahdr, 2025-07-17:
Maybe the converter is not loading? Are you seeing it being recognized as Moes device in the Manufacturer

Slejpnir, 2025-07-17:
Yes, the manufacture is Moes now.

fahdr, 2025-07-18:
After a lot of work and experimenting with the datapoints, the motor reversal does not work from zigbee2mqtt. I think this is set during calibration time (using the tuya app) and cannot be changed. Any help is appreciated

WRLPDZ, 2025-07-18:
> Most of the sensors work with this code. However i am not able to get the
> reversal and calibration to work
> e.enum('work_state', ea.STATE, ['standby', 'opening',
> 'closing']).withDescription('Current work state'),
> e.numeric('total_time', ea.STATE).withUnit('s').withDescription('Total
> operation time'),
> e.enum('situation_set', ea.STATE_SET, ['fully_open',
> 'fully_close']).withDescription('Situation control'),
> e.enum('fault', ea.STATE, ['none']).withDescription('Fault status'),
> e.enum('charging_status', ea.STATE, ['none', 'uncharged', 'charging',
> 'charged']).withDescription('Charging status'),
> e.numeric('open_threshold',
> ea.STATE_SET).withValueMin(0).withValueMax(100).withDescription('Light
> threshold for opening'),
> e.numeric('close_threshold',
> threshold for closing'),
> e.numeric('curtain_status',
> ea.STATE_SET).withValueMin(0).withValueMax(255).withDescription('Curtain
> e.numeric('total_distance', ea.STATE).withUnit('m').withDescription('Total
> distance traveled'),
> e.numeric('factory_test',
> ea.STATE).withValueMin(0).withValueMax(100).withDescription('Factory test
> feedback'),
> e.text('custom_week_prog_1', ea.STATE_SET).withDescription('Custom week
> program 1'),
> [... 33 lines cut ...]
> threshold (0-100)
> [110, 'curtain_status', tuya.valueConverter.raw], // Curtain status (0-255)
> [111, 'factory_test', tuya.valueConverter.raw], // Factory test (0-100)
> [112, 'total_distance', tuya.valueConverter.raw], // Total running
> distance in meters

Hi,

I have solution for now about reverse.

I have two of them on rood and one is reversed on rood. Then I have
calibrate it by button pressed.

BR,
PDZ

benlangfield, 2025-07-22:
Can anyone explain to me how to load the external converter?  Can't for the life of me get it to load...

resucutie, 2025-08-03:
Hello! Is the external converter functional (as in, the motors can close and open reliably, even if the open and close positions have to be set via the buttons)? Been wanting to buy these due to the low price and my low force requirement

matthewharrington, 2025-10-27:
Hi team.

Using @resucutie code above this device now works correctly for me. Any chance of implementing this @Koenkk

Koenkk, 2025-10-28:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

salphire3, 2025-11-18:
Hi, i use this code which work for me.

```
import fz from 'zigbee-herdsman-converters/converters/fromZigbee';
import tz from 'zigbee-herdsman-converters/converters/toZigbee';
import exposes from 'zigbee-herdsman-converters/lib/exposes';
import tuya from 'zigbee-herdsman-converters/lib/tuya';

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS030F', ['_TZ3210_sxtfesc6']),
    model: 'ADCBZI01',
    vendor: 'Moes',
    description: 'Curtain Robot',
    fromZigbee: [fz.cover_position_tilt, tuya.fz.datapoints],
    toZigbee: [tz.cover_position_tilt, tz.cover_state, tuya.tz.datapoints],
    exposes: [
        e.cover_position(),
        e.position(),
        e.battery(),
        e.illuminance(),
        e.enum('work_state', ea.STATE, ['standby', 'opening', 'closing']).withDescription('Current work state'),
        e.numeric('total_time', ea.STATE).withUnit('s').withDescription('Total operation time'),
        e.enum('situation_set', ea.STATE_SET, ['fully_open', 'fully_close']).withDescription('Situation control'),
        e.enum('fault', ea.STATE, ['none']).withDescription('Fault status'),
        e.enum('charging_status', ea.STATE, ['none', 'uncharged', 'charging', 'charged']).withDescription('Charging status'),
        e.numeric('open_threshold', ea.STATE_SET).withValueMin(0).withValueMax(100).withDescription('Light threshold for opening'),
        e.numeric('close_threshold', ea.STATE_SET).withValueMin(0).withValueMax(100).withDescription('Light threshold for closing'),
        e.numeric('curtain_status', ea.STATE_SET).withValueMin(0).withValueMax(255).withDescription('Curtain status'),
        e.numeric('total_distance', ea.STATE).withUnit('m').withDescription('Total distance traveled'),
        e.numeric('factory_test', ea.STATE).withValueMin(0).withValueMax(100).withDescription('Factory test feedback'),
        e.text('custom_week_prog_1', ea.STATE_SET).withDescription('Custom week program 1'),
        e.text('custom_week_prog_2', ea.STATE_SET).withDescription('Custom week program 2'),
        e.text('custom_week_prog_3', ea.STATE_SET).withDescription('Custom week program 3'),
        e.text('custom_week_prog_4', ea.STATE_SET).withDescription('Custom week program 4'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'open': 0, 'stop': 1, 'close': 2})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.coverPositionInverted],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': 0, 'opening': 1, 'closing': 2})],
            [10, 'total_time', tuya.valueConverter.raw],
            [11, 'situation_set', tuya.valueConverterBasic.lookup({'fully_open': 0, 'fully_close': 1})],
            [12, 'fault', tuya.valueConverterBasic.lookup({'none': 0})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'charging_status', tuya.valueConverterBasic.lookup({'none': 0, 'uncharged': 1, 'charging': 2, 'charged': 3})],
            [103, 'custom_week_prog_1', tuya.valueConverter.raw],
            [104, 'custom_week_prog_2', tuya.valueConverter.raw],
            [105, 'custom_week_prog_3', tuya.valueConverter.raw],
            [106, 'custom_week_prog_4', tuya.valueConverter.raw],
            [107, 'illuminance', tuya.valueConverter.raw],
            [108, 'open_threshold', tuya.valueConverter.raw],
            [109, 'close_threshold', tuya.valueConverter.raw],
            [110, 'curtain_status', tuya.valueConverter.raw],
            [111, 'factory_test', tuya.valueConverter.raw],
            [112, 'total_distance', tuya.valueConverter.raw],
        ],
    },
};

export default definition;
```

However, total distance do not work for me..and can anyone explain what, fully_set and curtain_status mean? Cannot figure it out.

resucutie, 2026-01-14:
> Hi team.
> 
> Using [@resucutie](https://github.com/resucutie) code above this device now works correctly for me. Any chance of implementing this [@Koenkk](https://github.com/Koenkk)

I wouldn't say this code is ready for production, mainly because of lacking features, and honestly i should do more testing with directionality (not sure if the device atm has issues with inverted directionality)

Also may a maintainer remove the stale mark?

jermbe, 2026-02-17:
I have installed this code, and I can open and close my curtain (although opening actually closes, the invert option doesn't change anything).
But my main issue is: how can I calibrate it, so that it knows when it is fully open/fully closed? Because the fully_open, and fully_close button do nothing. So now my curtain opens about 2/3rd of the way...

Any help is appreciated 😄

jermbe, 2026-02-18:
I also notice that on closing the motor always increases the distance with a couple of mm, so after 2-3 openings/closings it bumps into the wall.
I don't see any way on how to tweak this. It would be awesome if anyone could point me in the right direction to fix this. Thanks

hansons83, 2026-04-08:
@piano-thomas 
Hi. I tried fix the shaft with hammer and a pointy punch, it was tricky but it holds fine for few months now. I put some grease on the gears, it seems to improve the case when direction change and how gear engage (big gear pulls small one easier), but seems like grease make disengaging a bit worse when manually pulling the curtain to open/close it.

kingp0dd, 2026-09-05:
When i join device to z2m, it's being recognized as

Manufacturer
Tuya
Model
TS030F