Koenkk/zigbee2mqtt #24606: [New device support]: PIR + light sensor TS0601  _TZE200_gjldowo (ZM-35H-Q)

chevcheli0s, 2024-11-02:
### Link

https://www.zigbee2mqtt.io/devices/ZM-35H-Q.html

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c138b4b1f64a84","nwkAddr":43437,"manufId":4417,"manufName":"_TZE200_gjldowol","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId ":260,"epId":1,"devId":12,"inClusterList":[0,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stack Version":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1730565027993}

### Zigbee2MQTT version

1.41.0

### Comments

Please help me add support for this device.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_gjldowol',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing work

vantus1987, 2024-11-02:
Hi, try it, I set it up for myself today
https://github.com/Koenkk/zigbee2mqtt/issues/24085#issuecomment-2453123787

chevcheli0s, 2024-11-02:
> Hi, try it, I set it up for myself today [#24085 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/24085#issuecomment-2453123787)

Thanks!
I also tried this one https://github.com/zigpy/zha-device-handlers/issues/3061

chevcheli0s, 2024-11-04:
The same https://github.com/Koenkk/zigbee2mqtt/issues/24085

closed 2024-11-04 by chevcheli0s

BeeBerg, 2025-01-12:
I tried many different options, only this one worked. https://4pda.to/forum/index.php?showtopic=871505&view=findpost&p=133184799
the file itself
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_gjldowol',
        },
    ],
    model: 'TZE200_gjldowol',
    vendor: 'Tuya',
    description: 'Luminance motion sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.occupancy(),
            e.illuminance().withUnit('lx'),
            e.battery(),
            e
                .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
                .withDescription('PIR sensor sensitivity (refresh and update only while active)'),
            e
                .enum('keep_time', ea.STATE_SET, ['10', '30', '60', '120'])
                .withDescription('PIR keep time in seconds (refresh and update only while active)'),
            e
                .numeric('illuminance_interval', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(720)
                .withValueStep(1)
                .withUnit('minutes')
                .withDescription('Brightness acquisition interval (refresh and update only while active)'),
        ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse0],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [101, 'keep_time', tuya.valueConverterBasic.lookup({'10': tuya.enum(0), '30': tuya.enum(1), '60': tuya.enum(2), '120': tuya.enum(3)})],
            [12, 'illuminance', tuya.valueConverter.raw],
            [102, 'illuminance_interval', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;