Koenkk/zigbee2mqtt #24607: Device "Sonesse2 ULTRA 30 WF Li-ion Roll" is recognized but marked as "Not supported"

heisenberg2980, 2024-11-02:
### What happened?

I connected a blind from Somfy, model "Sonesse2 ULTRA 30 WF Li-ion Roll" and z2m recognized it and I am able to control the blind, but z2m shows the device as Not supported and some of the settings are incorrect. For instance, Home Assistant displays the status as unknown instead of open/close, but includes an attribute "current_position" with the correct position. Also Home Assistant display a Tilt option, but this is only a blind motor so it doesn´t have any option to tilt (this option should be removed from z2m)

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

20230507

### Adapter

SLZB-06

### Setup

Home Assistant Add-on

heisenberg2980, 2024-11-03:
After investigating a bit, I realised the logs in z2m indicate the blind reports the position and state in the variables **position_1** and **state_1**, and I guess z2m is using **position** and **state**, which are incorrect:

```
info 2024-11-03 15:01:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x4cc206fffe308590', payload '{"battery":99,"last_seen":"2024-11-03T15:01:51.153Z","linkquality":69,"position":0,"position_1":47,"state":null,"state_1":"OPEN","tilt":0}'
```

heisenberg2980, 2024-11-11:
@Koenkk do you know how I can update the definition of this Somfy blind motor so z2m picks the right values for position and state that the motor reports?

Koenkk, 2024-11-12:
Could you post the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition)?

heisenberg2980, 2024-11-12:
@Koenkk This is the generated external converter:

```
const {deviceEndpoints, windowCovering, battery} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Sonesse2 ULTRA 30 WF Li-ion Roll'],
    model: 'Sonesse2 ULTRA 30 WF Li-ion Roll',
    vendor: 'SOMFY',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"232":232}}), windowCovering({"controls":["lift","tilt"]}), battery()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

heisenberg2980, 2024-11-14:
@Koenkk YOU ARE A STAR, it works!!

And the only remaining bit was removing the tilt, but that was easy enough to do it myself, this is the final version:

```
const {deviceEndpoints, windowCovering, battery} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Sonesse2 ULTRA 30 WF Li-ion Roll'],
    model: 'Sonesse2 ULTRA 30 WF Li-ion Roll',
    vendor: 'SOMFY',
    description: 'Automatically generated definition',
    extend: [windowCovering({"controls":["lift"]}), battery()],
};

module.exports = definition;
```

I will follow the steps to add this to the z2m repo.

Just one last question, I followed the steps from https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html and I couldn´t figure this out. Is the knowledge you applied to fix this issue something you can write in a tutorial or some page with "other options to support new devices" so other people like me are "self-service" to add new devices? or is the knowledge and the possibilities too many to be able to write a comprehensive list of steps to follow?

closed 2024-11-14 by heisenberg2980

smoothquark, 2025-01-05:
Hi. I have the Somfy 24V blinds motors. I have set up external_converters in the /homeassistant/zigbee2mqtt/external_converters folder and it is picked up on zigbee2mqtt restart.

const {deviceEndpoints, windowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Sonesse 30 DC 24V Roller'],
    model: 'Sonesse 30 DC 24V Roller',
    vendor: 'SOMFY',
    description: 'Blinds',
    extend: [deviceEndpoints({"endpoints":{"1":1,"232":232}}), windowCovering({"controls":["lift"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition; 

They motors are now marked as "supported" and tilt no longer appears. However, state and position are still written to variables position_1 and state_1. The blinds work as expected within Home Assistant but Google Home integration falls over unless the I manually update the position and state variables. I presume I am missing a step somewhere?

Thank you.

smoothquark, 2025-01-05:
Oh. I copied and pasted my previous settings. I updated to follow your suggestion above:

const {deviceEndpoints, windowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Sonesse 30 DC 24V Roller'],
    model: 'Sonesse 30 DC 24V Roller',
    vendor: 'SOMFY',
    description: 'Blinds',
    extend: [windowCovering({"controls":["lift"]})],
};

module.exports = definition;

smoothquark, 2025-01-05:
Oh, it seems to be working now! Not sure why but it does and I can now control with Google Home. Fantastic!

Sytten, 2025-04-19:
I am wondering if others have that issue, but trying to use the blind makes it so the position is never fetched from the device so if you close using the command then the position is invalid. Inversely if you set the position (in a scene) to 0, then in Home Assistant the down buttons gets disabled permanently. It's fairly annoying.

heisenberg2980, 2025-04-19:
@Sytten I'm using the converter from https://github.com/Koenkk/zigbee2mqtt/issues/24607#issuecomment-2475753705, and the integration works perfectly, with the blind reporting the position, and I´m able to set the position from HA without any issues.

Sytten, 2025-04-21:
@smoothquark I pushed some PR to support our device officially https://github.com/Koenkk/zigbee-herdsman-converters/pull/9219 and https://github.com/Koenkk/zigbee2mqtt.io/pull/3718

kylelnsn, 2025-05-25:
BUMP

Sonesse2 ULTRA 30 WF Li-ion Roll still displays as "Not supported" but is usable. Also still showing tilt options which these motors do not have.