Koenkk/zigbee2mqtt #24610: TZE200_gjldowol

vantus1987, 2024-11-02:
### Link

https://aliexpress.ru/item/1005007626669999.html?spm=a2g2w.orderdetail.0.0.1b3b4aa6sR6xsM&sku_id=12000041560065090

### Database entry

{"id":58,"type":"EndDevice","ieeeAddr":"0xa4c13845da857e68","nwkAddr":47598,"manufId":4417,"manufName":"_TZE200_gjldowol","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":12,"inClusterList":[0,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1727184411211}

### Zigbee2MQTT version

1.41.0

### Comments

Good afternoon, I wrote an external converter, the sensor is working
https://github.com/Koenkk/zigbee2mqtt/issues/24085#issuecomment-2453123787

### External definition

```shell
const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_gjldowol',
        },
    ],
    model: 'TZE200_gjldowol',
    vendor: 'Tuya',
    description: 'Luminance motion sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.occupancy(),
            e.illuminance().withUnit('lx'),
            e.battery(),
            e
                .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
                .withDescription('PIR sensor sensitivity (refresh and update only while active)'),
            e
                .numeric('keep_time', ea.STATE_SET)
                .withValueMin(5)
                .withValueMax(3600)
                .withValueStep(1)
                .withUnit('seconds')
                .withDescription('PIR keep time in seconds (refresh and update only while active)'),
            e
                .numeric('illuminance_interval', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(720)
                .withValueStep(1)
                .withUnit('minutes')
                .withDescription('Brightness acquisition interval (refresh and update only while active)'),
        ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse0],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [12, 'illuminance', tuya.valueConverter.raw],
            [101, 'illuminance_interval', tuya.valueConverter.raw],
            [102, 'keep_time', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Everything works

zoomtronicOR, 2025-01-04:
There is no way I could make this PIR sensor becomes functional, I did update of the coordinator firmware,  created .js file in zigbee2mqtt folder in HA, an it does not work, device is still unsupported.

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_gjldowol',
        },
    ],
    model: 'TZE200_gjldowol',
    vendor: 'Tuya',
    description: 'Luminance motion sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.occupancy(),
            e.illuminance().withUnit('lx'),
            e.battery(),
            e
                .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
                .withDescription('PIR sensor sensitivity (refresh and update only while active)'),
            e
                .numeric('keep_time', ea.STATE_SET)
                .withValueMin(5)
                .withValueMax(3600)
                .withValueStep(1)
                .withUnit('seconds')
                .withDescription('PIR keep time in seconds (refresh and update only while active)'),
            e
                .numeric('illuminance_interval', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(720)
                .withValueStep(1)
                .withUnit('minutes')
                .withDescription('Brightness acquisition interval (refresh and update only while active)'),
        ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse0],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [12, 'illuminance', tuya.valueConverter.raw],
            [101, 'illuminance_interval', tuya.valueConverter.raw],
            [102, 'keep_time', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;`

rolfvreijdenberger, 2025-03-08:
thanks @vantus1987, through this post I could get my new pirs working in 2 minutes.  I created a pull request to incorporate the manufacturer https://github.com/Koenkk/zigbee-herdsman-converters/pull/8947

closed 2025-03-09 by Koenkk